opentx/opentx #8859: t-pro throttle cut on new model is assigned to a switch that doesn't exist

arkarkark, 2022-02-04:
**Describe the bug**
I made a new model and selected to add a throttle cut. I could not upload the config to my t-pro since the switch was not present.

**To Reproduce**
Steps to reproduce the behavior:
1. add a new model, I added a regular plane with a motor
2. select to add a throttle cut
3. try and upload to the t-pro tx
4. unable to do so.
5. go to mixes and change the switch to SA or SB    <---- trivial workaround
6. now you can upload to your tx

**Expected behavior**

Throttle cut should choose a default switch that exists.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version N638
 - Hardware Jumper T-Pro
 - OpenTX Companion OSX

arkarkark, 2022-02-04:
I'm very new to OpenTX, but I want to contribute, if you point me at the right file I'm happy to submit a PR for this. This seems like a good one to get started with helping with development.

kilrah, 2022-02-05:
Have a look here

https://github.com/opentx/opentx/blob/2.3/companion/src/wizarddata.cpp#L124