opentx/opentx #8860: Radiomaster Zorro Simulator on Linux Momentary switches missing

coaster007, 2022-02-05:
Edgetx companion/simulator on Linux: EdgeTx_Companion_2.7.0-44f12ba-x86_64.AppImage

Momentary switches SA/SD/SG not present in the Simulator, only the SH momentary switch
On Windows all momentary switches are available and visible.

[image]

rotorman, 2022-02-05:
If the issue is present solely in EdgeTX Companion, you might want to post the issue in EdgeTX GitHub instead: https://github.com/EdgeTX/edgetx/issues

coaster007, 2022-02-05:
Hi risto,
This issue is in simulator27 on linux.
But now an update ist only present on converted files from tx16s. If no covertion (new file new model) then the switches are present in simulator.
Greetings
Graziano

coaster007, 2022-02-06:
Sorry ! my fault posted in the wrong place !
sorry for the inconvenience, Graziano

closed 2022-02-06 by coaster007