opentx/opentx #8861: 6-position switch (buttons) initial value

arikrupnik, 2022-02-06:
I can't immediately see any discussion of this here or on the forums.

On my RadioMaster, the 6-position switch is always in position 1 when a model loads. Is there a way to set the initial position on this switch, persist it from the last time I used this model, or at least set a warning about its position like I can for toggle switches and pots?

I have a model where taking off with the 6P in position 1 would lead to a crash.

If there is already a feature request for this and I simply failed to find it, please accept my apologies.

3djc, 2022-02-07:
It is driven by external circuits, nothing OpenTX can do about it

3djc, 2022-02-07:
As for warnings; sevral way to do that:
- in model setup, activate and set pot and slider warning
- use special functions

closed 2022-02-07 by 3djc

rotorman, 2022-02-07:
@arikrupnik See Sec. 2.21 here: https://www.rcgroups.com/forums/showthread.php?3869543-Blog-17-RadioMaster-TX16S-schematic-diagram to understand why radio firmware cannot do anything about the 6-Pos. state (in short there is only a single unidirectional line from 6-pos to main radio microcontroller, physically there is no feedback line).

arikrupnik, 2022-02-07:
@rotorman Wow thank you for doing all this work documenting the hardware! This is a real service to the community. We need more people like you.

@3djc now that I understand the wiring, I'm ashamed to say I was looking in the wrong place for the warning. I didn't realize 6P was an analog input, so I was looking under switches. Now that I see that, it's obvious that the pot warning was there all along. Thank you for helping me gain a better understanding of the system.