opentx/opentx #8866: Mixers Disappear after Download of OTX File or During Simulation

Hartinger, 2022-02-13:
Yesterday I encountered  a weird problem with my Horus X10S Express running OpenTx 2.3.14.
One of the models uses some 'high mixes' for things like snap flap, etc.
I found that those high mixes have disappeared after downloading the model from Companion into the radio.
The mixers are not just invisible, they have no impact on the associated channels.
Furthermore, if I simulate the model in Companion, then those mixers disappear as well.

Interestingly, the mixers don't actually vanish, because if I reimport the OTX file into Companion, they show up again.
(As a test, I have recreated one of the mixers in the radio, and after reimporting the modified file into Companion, the mixer was duplicated.)

I have attached a test OTX file with just one model to illustrate the problem (it is about mixer lines 17 to 19).

[X10S_Test.zip](https://github.com/opentx/opentx/files/8055239/X10S_Test.zip)

bigtwin, 2022-02-13:
I could recreate the problem. I copied your _FunnyCub_ from your OTX and pasted it into my OTX file to prevent any unknow Radio settings.
The mixer lines 17, 18 and 19 where not visible in my _Horus 10S_ (2.3.14).  For a test I copied line 19 to mixer line 16 and it was also not visible.

I have not investigated that more in deep. Just as  quick thought ... could that be related to the fact that you bound with 8 channels? As one mixer line is linked to the according output, you may only have 8 mixer lines available? MAy worth some testing.

Hartinger, 2022-02-13:
I have tried with the latest nightly companion, same problem.

Hartinger, 2022-02-14:
I did some digging in my archives and found that the last file not exhibiting the problem is a 2.3.12 version from July 2021. The first file with the problem is from October 2021 and was archived using OTX 2.3.14. So something must have happened when I migrated from 2.3.12 to 2.3.14. (As I have no 2.3.13 archive I guess that I skipped that version, which was replaced by 2.3.14 after a short period).

I have attached the BEFORE and AFTER cases. I have deleted all but two models that actually exhibit the problem. Maybe this helps to find out if/how the file got corrupted, or what the problem is with Companion 2.3.14.

[X10S_Test_Jul_2021_OTX_2_3_12.zip](https://github.com/opentx/opentx/files/8058211/X10S_Test_Jul_2021_OTX_2_3_12.zip)
[X10S_Test_Oct_2021_OTX_2_3_14.zip](https://github.com/opentx/opentx/files/8058212/X10S_Test_Oct_2021_OTX_2_3_14.zip)

elecpower, 2022-02-14:
In Companion reading the X10_Test.zip:
- if you copy and paste any of the vanishing lines above or below then run the simulator the copied lines are displayed in the simulator and are editable. Whereas the vanishing do just that as if you try to edit in the simulator they are defaults;
- copy and paste all lines to blank channels, delete the vanishing lines, move the copied lines back to their original position and amazingly they show and are editable in the simulator.

So at this time I am leaning towards a subtle corruption. It is a strangle one as one would expect any copied lines to also exhibit the same issue but they don't.

Hartinger, 2022-02-14:
I will bin the file, which might indeed be corrupted, and use the latest archive that didn't have the problem. As you said, it is a strange one, as I did not do anything special. I just migrated the OTX file from 3.12 to 3.14 without any obvious issues. And there are other models that use an auxiliary mixer at channel 17 and have no issues.

RC-SOAR, 2022-03-03:
I've experienced something like this.

As I recall, if the source of a mix is defined in Companion but undefined in the actual hardware, it leads to problems - after transfer to the hardware, all subsequent mixers are hidden/ignored.

Hartinger, 2022-03-04:
The problem was related to the following mixes, so I don't beleive there was a problem with the sources.

[image]