opentx/opentx #8869: Suggestion: resetting a variable (e.g. "var") shouldn't automatically reset "var+" and "var-"

aeronaut999, 2022-02-24:
Suggestion-- when any given variable such as "Alt" is "Reset", the "+" and "-" versions of that variable (e.g. "Alt+" and "Alt-")  should _not_ automatically be reset.

This would avoid the need to write special programming lines to remember the maximum (or minimum) value of a variable that may be reset several times during a flight.

If there is a special reason to reset the "+" and/or "-" versions of the variable, that can always be accomplished with one or two extra programming lines in the "Special Functions" area.

(Tangentially related to this issue -- https://github.com/opentx/opentx/issues/7949 )

bigtwin, 2022-02-24:
I am curios and its not really clear to me what this is about ...
What value would I want to reset if not a maximum of e.g. _Alt_ or the minimum of e.g. _VFAS_? I am doing this rather often using "_Reset Telemetry_". 
How is this related to your suggestion?

bigtwin, 2022-02-24:
Wow! Thx for the example/ explanation. I never heard of something like that. And actually I dont not see a need for such a sophisticated config on my planes.

kilrah, 2022-02-24:
You can duplicate Alt3 to a new sensor that you don't reset, it'll then keep the +/- and you just don't look at the value itself.

aeronaut999, 2022-02-24:
Kilrah, I don't see any way to duplicate a sensor in a way that would cause it to remember the value 1000' in the example given above.  Do you?  If so please explain more, because I am missing something.

Of course it is simple to duplicate a sensor that would end up remembering the value 1400' in the example given above.  Simply zero out the duplicate sensor before launch, and never adjust it.  I've already done that.

aeronaut999, 2022-02-26:
Re -- 

"New sensor (call it "alt4") is a "calculated" sensor
Formula is "add"
Source is "alt3" 
Then no matter how many times I reset "alt3", "alt4+" will be the maximum value that alt 3 ever reached.

Would that work?  Is the program ok with having only one source for an "add" calculation?"

Works perfectly, mission accomplished.  
(Actually I used formula "max" rather than "add", but "add" would have surely given the same result)

Suggestion withdrawn --

kilrah, 2022-02-26:
Cool.

> (Actually I used formula "max" rather than "add", but "add" would have surely given the same result. The point is simply to make "alt4" be always equal to alt3.

Simply duplicatting Alt3 would have achieved the same result, hence being what I initially suggested.

Given the complexity of that setup it seems more appropriate for a lua script though, where you could make a custom display that shows the things you care about in a more convenient way, don't know if it's the case for your particular use but there are people who have made that kind of thing for glider stats long ago already.

kilrah, 2022-02-26:
OK, I guess you're using the auto offset, which I missed.

aeronaut999, 2022-02-27:
> OK, I guess you're using the auto offset, which I missed.

I think there's more to it than that, it has to do with intention to reset the variables whenever the values go negative.   To keep track of the climb from the lowest low point following the end of the latest motor run.

3djc, 2022-02-28:
Really this scenario would be sooo much better suited for LUA, you could really make some thing very nice using that

aeronaut999, 2022-02-28:
> Really this scenario would be sooo much better suited for LUA, you could really make some thing very nice using that

It is working perfectly fine this way too-- I have yet to dive into writing and using LUA scripts--

jrwieland, 2022-02-28:
> > Really this scenario would be sooo much better suited for LUA, you could really make some thing very nice using that
> 
> It is working perfectly fine this way too-- I have yet to dive into writing and using LUA scripts-- and am kind of afraid to start because I have a whole bunch of ideas for other little projects I'd want to do, would take up all my flying time!
> 
> Anyway, suggestion withdrawn--

I have free time and dumb enough to offer to write a script, based on your input, to get you the information on the screen that your after.  With a script you can  have a chart showing each of your accents plus any other items that you would desire.  Literally,  it can be designed to put anything on the screen.  here is a screenshot of the one that I have for my helicopters (food for thought).  Just let me know if your interested
[image]