opentx/opentx #8874: Turning on internal MPM screws the synchronization with the external ELRS module

drtrigon, 2022-03-07:
**Describe the bug**
When using an external ELRS module it is not possible to use the internal MPM as wireless trainer link. Consider e.g. following setting:

`use the external ELRS as link to a model and the internal MPM as wireless trainer link to a spectrum transmitter`

As discussed on discord with @raphaelcoeffic (pafleraf) and @3djc, currently "you cannot synchronize on 2 modules at the same time. At least not if it triggers the mixer as now. The current implementation synchronizes on the internal module if ON, and on the external otherwise." "The fact that the internal MPM is turned on screws the synchronization with the external ELRS module!" "Ideally, we should disable the synchronization on the MPM when it works as a RX." "That should be enough to overcome the issue."

**To Reproduce**
Steps to reproduce the behavior:
1. use ELRS as external module
2. https://www.expresslrs.org/2.0/quick-start/transmitters/tx-prep/ states at multiple places "NOTE: Make sure the internal module is turned off."
3. so the internal MPM cannot be used at all

**Expected behavior**
It should be possible to use the internal MPM even when there is an ELRS external module present and used (active).

**Screenshots**
(none)

**General information**
 - OpenTX version: 2.3.14
 - Hardware: Radiomaster TX16S
 - OpenTX Companion: -
 - (ELRS: 2.2.0)

drtrigon, 2022-03-07:
https://github.com/EdgeTX/edgetx/issues/1671

pfeerick, 2022-03-26:
As far as how we've implemented this in ETX, I think we need to dig into this more, as at the moment it is still primarily dependent on a hacky way of setting `status.isRXProto`... which will backfire quite nicely if a protocol name ever ends with "RX". But it is functional enough at the moment to get by until either another bit of info is relayed from the MPM, or a more reliable mechanism is implemented.

3djc, 2022-03-26:
I'll try to lock at required MPM changes over the week end, if I can find the time for that

3djc, 2022-03-27:
Had a long chat with @raphaelcoeffic , and we both agree that this is just the tip of the iceberg, and ignores most of the real issue (ie if you want to use internal MPM TX + ext ELRS, it will faill). It all boils down to managing the priority of modules, and we havent found the perfect answer just yet

drtrigon, 2022-03-27:
I agree - I also came up with several configurations that might cause issues.

It might not be possible to do all this automatically and finally it thus might be needed to introduce a param that can be setby user.

However I really would like to have the option to use the internal MPM as trainer link, e.g. for external elrs, crossfire, r9m, etc. Any chance that will come to otx soon?

3djc, 2022-03-27:
It does work right ? All you loose it the ability to sync to module (which does not exist at all for r9m), which in the scenario of wireless training is a bit pointless too, no ?

drtrigon, 2022-03-27:
Obviously I did miss something - so what is the difference between beeing synced and not? (if it works w/o beeing synced, what is the advantage of syncing?)

3djc, 2022-03-28:
Syncing helps a little betaflight computation. Whenever you can feel it on craft control is a hightly polemical subject  we rather don't want to debate here.

raphaelcoeffic, 2022-03-29:
In a nutshell, the sync is about latency and input aliasing. At least, this is what it solves.

regarding the scenario here at hand, what happens is that you will have a slower update frequency of inputs (150Hz vs more) and possibly some small aliasing.

drtrigon, 2022-03-29:
(I see ... thanks for the explanaition ... from reading the ELRS quick starting guide I got the impression that this sync is absolutely mantatory otherwise the universe might break ... ;) but this is actually something to be updated on their wiki and not relevant here)