opentx/opentx #8877: Getting Assert on Companion23 and Simulator23 after commit 59f89a924a296f366c07b7729bc1ab8a92ec0078

jtaylor2, 2022-03-12:
I am getting an ASSERT when running my self built Companion23 or Simulator23 after this mornings commit 59f89a924a296f366c07b7729bc1ab8a92ec0078

ASSERT: "rd.maxSubtype + 1 == (unsigned int) rd.protocols.length()" in file /root/frsky/opentx/git/opentx/companion/src/firmwares/multiprotocols.h, line 65
Aborted (core dumped)

Latest 2.3 branch compiled on openSUSE 15.3

3djc, 2022-03-12:
@MRC3742 did you miss something, like Xerall that is not in all lists ?

MRC3742, 2022-03-12:
@3djc Yes, I forgot to add the string in Companion for Xerall and MT99XX2.
Will add them in a new PR later today.

3djc, 2022-03-12:
Cool thanks

MRC3742, 2022-03-12:
@3djc  @jtaylor2  #8878  should resolve this "Issue"

Rich

jtaylor2, 2022-03-13:
I'm still the core dump and assert on both Companion and standalone simulator.
Where is the value of rd.maxSubtype set?  I can't find it?

3djc, 2022-03-13:
When exactly do you get those?

jtaylor2, 2022-03-13:
As soon as I try to run them.  Are you able to run the current 2.3 branch?

ASUS-Linux:~/frsky/opentx/git/opentx/build-test # ./simulator23
ASSERT: "rd.maxSubtype + 1 == (unsigned int) rd.protocols.length()" in file /root/frsky/opentx/git/opentx/companion/src/firmwares/multiprotocols.h, line 65
Aborted (core dumped)

3djc, 2022-03-13:
Yes, I can run companion just fine, tried selecting a multi module, and select the new protocols, worked fine

jtaylor2, 2022-03-13:
Must be something with my environment.  I build almost every day and was working fine until that commit.  Will try to see what's wrong with my build environment.

3djc, 2022-03-13:
Just pushed a consistency change, does it improve something for you ?

jtaylor2, 2022-03-13:
Still doesn't work.  It works fine if i comment out that assert.  I don't understand the code enough to know where rd.maxSubtype gets set or what value it should hold.  Maybe something with my options.  I'll create a clean build directory and try with no options.

3djc, 2022-03-13:
maybe share what you exactly build so we can try to reproduce

jtaylor2, 2022-03-13:
cmake -Wno-dev -DCMAKE_BUILD_TYPE=Debug -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DEXT=FRSKY -DSDCARD=YES -DPPM_CENTER_ADJUSTABLE=YES -DFLIGHT_MODES=YES -DAUTOSWITCH=YES -DPPM_LIMITS_SYMETRICAL=YES -DAUTOSOURCE=YES -DDBLKEYS=YES -DHELI=YES  -DPPM_UNIT=PERCENT_PREC1 -DGVARS=YES -DFAI=NO -DUNITS=IMPERIAL -DTRANSLATIONS=EN -DHAPTIC=YES -DLUA=YES -DOVERRIDE_CHANNEL_FUNCTION=NO -DMIXERS_MONITOR=YES -DTIMERS=3 -DMULTIMODULE=YES -DALLOW_NIGHTLY_BUILDS=NO -DSIMU_AUDIO=YES ..

Trying now with just PCB=X10 and PCBREV=TX16S

jtaylor2, 2022-03-13:
Build with just  PCB=X10 and PCBREV=TX16S works.  Something with my options.  I haven't changed them in years.  Will clean them up to just what I need now and if it still fails figure out which one does it.

kilrah, 2022-03-13:
> cmake -Wno-dev -DCMAKE_BUILD_TYPE=Debug

Looks like Release build is fine, error only shows up on a Debug build.

3djc, 2022-03-13:
Fixed

jtaylor2, 2022-03-13:
Missing coma.....guess that's why asserts are there.  Thanks

closed 2022-03-13 by jtaylor2