opentx/opentx #8890: Companion freezes when the simulator is closed

MacFlieger, 2022-03-21:
I have a sound file SYSTEM/bye.wav that normally plays when the transmitter is turned off. This works.
When I exit the simulator with this file present in Companion, Companion immediately and reproducibly freezes and can only be hard-quit. If the file is not present, then exiting the simulator works normally.

To reproduce:
1. Put a sound file named bye.wav and the sound folder SYSTEM.
2. Start the simulator in Companion.
3. Exit the simulator

Versions:
- OpenTX Companion 2.3.14, but also tested 2.3.15 nightly
- macOS Big Sur 11.6.5

MacFlieger, 2022-09-21:
The problem still exists, even with actual versions:
OpenTX Companion 2.3.15
macOS Monterey 12.6

bigtwin, 2022-09-21:
Will that happenes with every bye.wav? Can you provide your bye.wav?

MacFlieger, 2022-09-22:
Now I have tested with an older computer:
Companion 2.3.15 with macOS High Sierra 10.13.6
"Simulate Radio" in model selection main window: Closing simulator works as expected.
"Simulation" in model config window: Closing simulator closes the model config window immediately too, but no freeze of the program.
In both situations the sound "bye.wav" ist not played, but I think this is normal and it is no problem.

Tested with newer computer (x86):
Companion 2.3.15 with macOS Monterey 12.6
"Simulate Radio" in model selection main window: Closing simulator freezes the program.
"Simulation" in model config window: Closing simulator freezes the program.

Probably it is only a problem of companion with newer macOS versions?

bigtwin, 2022-09-22:
Ok, one more reason to not upgrade early. Actually better to stay backleveled for a couple of years. ;-) Anytime MAC OSX upgrades I run into quite some issues :-( Not that bad as it is on Windows but it is still annoying.

I feel like Companions version for MAC OSX can not be maintained that good due to lack of ressources. Have you checked the EdgeTX branch of Companion?

MacFlieger, 2022-09-22:
> Ok, one more reason to not upgrade early. Actually better to stay backleveled for a couple of years. ;-) Anytime MAC OSX upgrades I run into quite some issues :-(

I agree mostly.
But at some point, newer software isn't working on older macOS, because of the use of newer APIs.
Another problem with an old macOS version I had: Not long ago a version of OpenTX Companion doesn't work on macOS High Sierra 10.13 and older. The nightly builds worked, but not the release ones. The only advice I get: Update macOS, but it wasn't possible on my old mac.
After some research I found the difference in the building scripts for the nightly an release versions and the bug was patched.

I haven't checked EdgeTX yet. For my purposes there is no advantage over OpenTX.