opentx/opentx #8893: controling TX16s using uart

ubairsofi, 2022-03-30:
Hello 
I am working on OPenTX (model : TX16s) . I want to control the drone through the GUI for that i have to send data through uart port.
I need to know is it possible to control all the switches from the uart port and  if i can control what are commands.

raphaelcoeffic, 2022-03-30:
One way to do it would be to implement S.BUS on the host side and use the trainer input. That is, if this is about controlling a model.

rotorman, 2022-03-30:
@ubairsofi have a look at https://doc.open-tx.org/opentx-2-3-lua-reference-guide/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/serialwrite to output data from AUX1 or AUX2 of TX16S from LUA. With EdgeTX you could even have touch-screen support from a LUA widget outputting data from AUX1 or AUX2. See an example for this: https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page932#post48855127 (here AUX2 is used internally to drive the PWM controller from a LUA widget).