opentx/opentx #8897: STM32__BOOTLOADER driver install and Windows 10 fails consistantly

Flyswadder, 2022-04-03:
STM32 Bootloader driver consistanly fails Windows 10 install on multiple machines.

[image]
[image]

3djc, 2022-04-03:
Sorry but what you experience is a local pc issue, not an OpenTX bug

closed 2022-04-03 by 3djc

Flyswadder, 2022-04-03:
It's your driver that refuses to install on ANY windows 10 machine, multiple computers, always the same failure. How is that a "local" pc issues when it doesn't matter what Windows 10 machines, in multiple locations across the country?

3djc, 2022-04-03:
What you call our driver is stm32 bootloader, from st electronics, and has nothing to do with OpenTX. If I might add, it used by millions of devices worldwide

rotorman, 2022-04-03:
@Flyswadder where did you get that driver? For the official ST DFU drivers, download STM32CubeProgrammer, that comes packaged with the original DFU drivers (you need to make a free account at ST to download it); https://www.st.com/en/development-tools/stm32cubeprog.html

kilrah, 2022-04-03:
1) You shouldn't need them in situations other than recovery from a "bricked" radio
2) You use Zadig to install them

Flyswadder, 2022-04-03:
This is the source of that file Rotorman: https://www.open-tx.org/2013/07/18/taranis-windows-usb-driver-for-manual-installation.

I have since downloaded the cube file from ST Electronics and it installed, but I'm now not sure what to do with it. I managed to update my bootloader and firmware from an obsolete Win 7 machine that the above software installed properly on.