opentx/opentx #8898: Differenz values, using P6_1 till P6_6 (chicken feet)

kurtreit, 2022-04-07:
It is not possible to create logical switches, using the chicken feet (P6), because companion generates different outputs for P6_1 till P6_6

jrwieland, 2022-04-07:
I just checked that nightly (Because I do use the "chicken feet" for my flight modes). Initially the values were off when I tried to set them

kurtreit, 2022-04-07:
I used the radio profile Jumper T18. There is no chance to get predictable results. P6 is assigned to Channel 10.
[image]
Should be -60%
[image]
Should be -100%

bigtwin, 2022-04-07:
I can confirm. 2.3.14 on MAC OSX. The values for each position of 6P differ heavily in _Companion_. I noticed that the value for the switch seems to be "triggered" my the mouse movement/position rather than by the actual switch position indicator (the little, hard-to-tell triangle).

kurtreit, 2022-04-07:
@bigtwin 
Yes, you are in right! I think they should rework the snapping position of the mouse cursor for exact assignment.

kurtreit, 2022-04-07:
@jrwieland 
With the wheel, its fine