opentx/opentx #8899: XJT External module not working with OpenTX 2.3.14

adop-adop, 2022-04-07:
Hello Everyone

I have an issue with my transmitter module XJT installed on Jumper T18 and running under OpenTX 2.3.14
The external module XJT not working with the OpenTX 2.3.14 

I confirm my module is perfectly functional I have test him on friend's X7 Taranis, all working perfectly 

And this XJT module have already function on my Jumper T18 with an older OpenTX Version 
the issue is that the XJT not entering in binding
both LED (green and red) stay fixe 
for compare with the X7 taranis the green LED blink rapidly and red is fixe (except for the bind the red flashing)

Can any one have the same issue ? and find a solution ?
I precise that my firmware is FCC and my opentx too 
I have also try to flash different firmware on the XJT nothing working

thanks

paulo-sousa, 2022-05-25:
I likely have the same issue. After upgrading from OpenTX 2.3.10 to **OpenTX 2.3.15**, the **Internal XJT** module on my **Taranis Q X7** stopped working correctly. The firmware on the XJT is 170317.

I tried 2 models with receivers in D8 mode and the radio stopped connecting to both.

If I perform a "Bind" operation, it works correctly (tested on both models). Immediately after binding, if I don't reboot the radio, the connection is established correctly. If I reboot the radio, the connection is no longer established.

If I flash the radio back to OpenTX 2.3.10, the connection starts working immediately, without the need to rebind. Upgrading to OpenTX 2.3.15 consistently makes it stop working again.

It is as if the radio in OpenTX 2.3.15 is not sending the correct ID to the XJT when starting up or changing models, but keeps the ID after the bind operation.

paulo-sousa, 2022-05-25:
I've tried with a receiver in D16 mode and it this mode works correctly. Also tried to narrow down the specific version where the issue first appeared, but I could not find a way to download an older firmware version.

This seems to isolate the issue to **"XJT Module does not connect to receiver in D8 mode"**.

Additional notes:
 - D8 bind operation still works correctly. Only the standard connection does not work.
 - Issue appeared between OpenTX 2.3.10 and OpenTX 2.3.15.

kilrah, 2022-05-26:
What's the name of the 2.3.15 firmware file you're flashing?

paulo-sousa, 2022-05-26:
> What's the name of the 2.3.15 firmware file you're flashing?
I am flashing the latest version downloaded by **Companion** for Taranis Q X7, with the default radio settings.
The file name is `opentx-x7-en-2.3.15-otx.bin`.

alm-nl, 2022-05-31:
The issue is related to ACCST v1 and it does not happen in ACCST v2.1 as can be read here: https://www.rcgroups.com/forums/showthread.php?3961323-QX7-wont-bind-to-D8-receivers-after-OpenTX-upgrade/page2