opentx/opentx #8901: add support touchscreen and refactor GUI interface

cvetaevvitaliy, 2022-04-11:
Full refactor GUI interface with LVGL https://github.com/lvgl/lvgl
This will solve the problem full touch GUI support
This GUI library is very functional, has excellent animation and many elements: buttons, sliders, sliders, lists, menus and more

rotorman, 2022-04-11:
LVGL is WiP in EdgeTX already, see: https://github.com/EdgeTX/edgetx/projects/12 and https://github.com/EdgeTX/edgetx/tree/lvgl

cvetaevvitaliy, 2022-04-13:
> LVGL is WiP in EdgeTX already, see: https://github.com/EdgeTX/edgetx/projects/12 and https://github.com/EdgeTX/edgetx/tree/lvgl

Thanks for reply 

I don’t  see sources LVGL project
I see submodule and custom GUI https://github.com/EdgeTX/libopenui  in EdgeTX 

I think this giid idea for refactor GUI OpenTX firmware

rotorman, 2022-04-13:
> I don’t see sources LVGL project

LVGL is a linked as a submodule in libopenui: https://github.com/EdgeTX/libopenui/blob/f888bc724a0e858ac3da3c7cd97031b7ee8fb6b0/.gitmodules#L4-L7