opentx/opentx #8909: OTX 2.3.15 crashes when loading OpenTX-Telemetry-Widget

Arvycka, 2022-04-25:
OTX 2.3.15 crashes into emergency mode when loading [OpenTX-Telemetry-Widget](https://github.com/iNavFlight/OpenTX-Telemetry-Widget)

https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/40

**To Reproduce**
OpenTX-Telemetry-Widget load normaly in OTX 2.3.14
Update OTX to 2.3.15
a. Switch screen to telemetry widget - OTX crashes.
b. Add new screen, select telemetry widget - OTX crashes.

**General information**
 - OpenTX version 2.3.15
 - Hardware Radiomaster TX16S

cchaz003, 2022-04-30:
I'm running into this exact same issue as well. Any help or guidance would be welcome

robertw2a, 2022-04-30:
I have the same issue on an X10S radeo worked fine until i selected a model using INAV telemetry lua. Had to switch back to 2.3.14 to be able to change models and use INAV telemetry lua

cchaz003, 2022-05-01:
I wish I had a past version to fall back on. I just got the TX16S so this is my first version on here. @Arvycka do you happen to have a 2.3.14 version for the TX15S that I could use for now?

bigtwin, 2022-05-01:
Usually you get former versions thru the common support forums. But also usually such major bugs are fixed rather fast.

robustini, 2022-05-01:
Same issue here with Horus X10S and Radiomaster TX16S.

3djc, 2022-05-01:
The thing is we are not the right folks to troubleshoot this. We need those widget developer to investigate and find the cause. If it turns out to be an OpenTX issue, then they can tell us : when we make this call, it crashes

stronnag, 2022-05-01:
The same script works in OpenTX 2.3.14 and earlier, it works in EdgeTX; the only problem is OTX 2.3.15.
So let me guess, there are undocumented API changes ... 
Yep. Ok we'll update the widget.

3djc, 2022-05-01:
Maybe there is, wanted or not, if we are doing somethign wrong, happy to fix, just need a more specific pointer to what the issue is

stronnag, 2022-05-01:
The issue is that the change the `getVersion()` to addtionally return `osname` is not documented, neither in the [release information](https://www.open-tx.org/2022/04/22/opentx-2.3.15) , nor in the [Lua guide](https://doc.open-tx.org/opentx-2-3-lua-reference-guide/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/getversion).

Resulting in lazy checks for `osname ~= nil` to distinguish OTX / ETX failing.

3djc, 2022-05-01:
good point, lua guide need to be nanually rerun

robustini, 2022-05-01:
> The issue is that the change the `getVersion()` to addtionally return `osname` is not documented, neither in the [release information](https://www.open-tx.org/2022/04/22/opentx-2.3.15) , nor in the [...]

Did you fix it on your repo?

stronnag, 2022-05-01:
> Did you fix it on your repo?

https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/40#issuecomment-1114181428

3djc, 2022-05-01:
Closing then, taken a note to amend release notes and trigger a lua doc refresh

closed 2022-05-01 by 3djc