opentx/opentx #8911: On Taranis radio screens, the exclamation mark indicating emergency reboot is displayed as an inverse character.

Miami-Mike, 2022-04-30:
This one should be easy to fix. [This Taranis X9 Lite screencap](https://www.rcgroups.com/forums/showthread.php?p=49092605&postcount=8061) and [this Q-X7 screencap](https://www.rcgroups.com/forums/showthread.php?p=49099851&postcount=8075) show what we on RCGroups determined to be the exclamation mark (!) emergency mode warning icon, but displayed in inverse against a light background resulting in an unrecognizable symbol. 

https://static.rcgroups.net/forums/attachments/8/7/4/9/1/5/a16105071-253-f2.png

https://static.rcgroups.net/forums/attachments/1/6/2/4/9/a16109289-103-B88CB180-FB8B-47F0-9E45-4CEE4AFBA0C4.jpeg

kilrah, 2022-04-30:
Always been so, little more obvious that way.

closed 2022-05-03 by 3djc

Miami-Mike, 2022-05-03:
Are you going to fix this? Please look at the images and RCGroups discussion that I linked to. The symbol **would** look like an outlined exclamation mark if the top line was visible on 128 x 64 screens, but if that's what was intended then you need to move it down one row. What shows looks sort of like an inverted capital "A".