opentx/opentx #8912: [LUA] How to use foreign fonts in LUA scripts?

LapinFou, 2022-05-02:
Sorry for the silly question, but I can't find any relevant information.
Look like the special foreign accents are available: [](https://doc.open-tx.org/opentx-2-3-lua-reference-guide/part_vii_-_appendix/fonts#french)
However I don't understand how to use it, for example, with `lcd.drawText` function.

3djc, 2022-05-03:
You cannot,it would require some kind of Unicode support, and that would require much more ressources

kilrah, 2022-05-03:
You can use them with an escape sequence
`"T\128l\128commande:"`

3djc, 2022-05-03:
but that only works with matching firmware language (if using english firmware, you'll get garbage

LapinFou, 2022-05-05:
Thanks for your answers. To keep it simple, I will stick with the "no accent" solution.

closed 2022-05-05 by LapinFou