opentx/opentx #8916: plausibility check for telemetry values

bigtwin, 2022-05-09:
It would be nice if OTX can check the plausibility of a telemetry value, before processing it.

Sample:
Once in while it happens that the distance or heigh value from my GPS sensor (FrSky) is showing useless values. Instead of some like distance like e.g. 312m, suddenly I have like 
12371m or 20629m displayed. Just some strange value which probably occured due to a transmission failure. And now such a value make it into Maximum/Minimum scores and as well into the logs.
And all happens while the telemetry basically continue to work fine, including GPS distances.

In reality it will never happen that a reasonable distance change is more than a couple of 10m among a measurement period. OTX may can check whether a telemetry value makes sense at all and may even in comparison to the last value(s). If a value is not sensefull, it can simple be dropped and does not pollute logs and statistics.

3djc, 2022-05-09:
We do check for telemetry paquets CRC, so it should handled transmission errors. Now if sensor send crap, thats likely where it needs fixing

bigtwin, 2022-05-10:
Sorry, I am not writing about fixing. Its rather some workaround. It would surely be better to catch the root cause for those senseless telemetrie values. However, if OTX displays a telemetry value of 25.754m distance it is simply clear that something went wrong. Especially if the value before was 315m. So CRC check is not catching that for whatever reason.
Of cause OTX is not the problem here but OTX would be in the position to simply cut such a senseless value out and ignore it.

So plz take it as suggestion or trash it. I am not able to code it and all I can contribute is the idea.

Rotoryfr, 2023-05-02:
Hi,
I regularly have wrong results in distance. When i check the log, GPS position received is correct, but the distance is wrong, generally a high value, 40.000m yesterday. I think there is a bug in distance calculation. All my models are equipped with altimeter and GPS, and I notices this problem with my X9D and my X12S on several models.
In the attached file, wrong distances are yellow, the last two columns show the correct distances (with and without altitude) calculated through Excel. You can see that GPS coordinates are coherent, so the problem is in distance calculation.
[Extrait et recalcul DRAGON-2022-06-04.xlsx](https://github.com/opentx/opentx/files/11370075/Extrait.et.recalcul.DRAGON-2022-06-04.xlsx)

bigtwin, 2023-10-19:
I have started this issue, but I am not anymore interested in a solution. Since @Rotoryfr provided a contribution, I leave it up to him to keep the issue open or close it.

Rotoryfr, 2023-10-19:
The file I joined clearly shows that the problem is not the sensor sending crap, but miscalculation. Please have a look at this problem.