opentx/opentx #8922: Stuck in the menu with "Failsafe not set"

mklueh, 2022-05-24:
Hello,

I'm facing a problem with my Radiomaster TX16s. When booting, I'm stuck in with the prompt "Failsafe not set [RTN]"

But no buttonpress will make this window disappear. I have even bought a new button platine and tried it again, with no impact. The first two windows can be closed with the return button but this one is not going away.

What can I do now? Flashing firmware with the sd card and without connecting the device is not possible, right? But connecting with USB does not work either.

The Firmware I'm having is the last one from last summer.

mklueh, 2022-05-24:
Ok so I've created a new empty model with the companion and set it as default. Now I'm getting past that popup finally, but it seems the back button is not doing what it should do

3djc, 2022-05-24:
You can check buttons in SYS/Hardware/Keys (Bottom of hardware screen)

mklueh, 2022-05-25:
@3djc thanks, so according to the debug view the return button is working correctly, but the return action is not performed

3djc, 2022-05-25:
Hardware issue then

closed 2022-05-25 by 3djc

mklueh, 2022-05-25:
@3djc how if the signal seems to be correctly interpreted?

3djc, 2022-05-25:
Sorry, I miss read you message. Anyhow, given the number of tx16s, if retrun wasn't working, we would know it. Try to flash 2.3.15 maybe

mklueh, 2022-05-25:
@3djc thanks. I've noticed that when I press return also t4 is triggered, but with the latest version (now tried edgeTX) I'm at least not stuck anymore in the screen and can keep using my remote.