opentx/opentx #8923: Extended limits limited to 150. Brain FBL no longer configurable.

ReneHitz, 2022-05-24:
In the current firmware for the radiomaster TX16S, the servo travel can be set to a maximum of 150%. This is not sufficient to control a Brain FBL controller. The internal value in the brain then only just goes up to 86% instead of 100%. OpenTX 2.3.15 with internal multi module 1.3.3.7 are installed.

kilrah, 2022-05-26:
150% in OpenTX is more than any system puts out so that's unlikely to be the problem, if anything you're probably giving too much travel, not too little. Also that is so much that most modules/receivers will not accept it and may either truncate it or even turn themselves off because of the signal being considered invalid.

mawzthefinn, 2023-02-26:
Using Sbus to the FBL? Then you are limited to 125% as a maximum valid range. The Sbus protocol does not support output ranges greater than 125%.

mawzthefinn, 2023-02-27:
The reason you have it in EdgeTX too is that problem has nothing to do with OpenTX or EdgeTX, it's a limitation of the SBus protocol. 

The maximum channel limits allowed in SBus equate to +/- 125% in OpenTX, EdgeTX or related OS's (FrOS and ETHOS share the same channel range limits)

MSH can't fix it either, since it's Futaba's fault.

ReneHitz, 2023-02-27:
The only question is whether you can still fly safely with this configuration. Other members of my club have the same problem and also fly with this configuration. I actually wanted MSH to make a statement on this, but they don't seem to want to do that.

ReneHitz, 2023-02-27:
I will probably connect a standard receiver and configure the heli with it. Unfortunately, it's a lot bigger than a simple S-Bus receiver.

3djc, 2023-02-27:
Isn't the reasonable solution to simply follow the standards and use 125% ?

ReneHitz, 2023-02-27:
This does not achieve the required signal ranges in the configuration

3djc, 2023-02-27:
What signal range ? You mean servo range ? Surety the flybarless unit has configurable gains, or it can be mechanically increased with longer harms

ReneHitz, 2023-02-27:
I already know that I can (or even have to) correct it mechanically. The servo travels are actually much too large at the end. Since we are still with the same problem: Is the configuration so safe?

3djc, 2023-02-27:
Using 150% with SBus is not safe, no

ReneHitz, 2023-02-27:
Ultimately, this means that the S-Bus can no longer be used with the Brain. It might be time for MSH to confirm that.

kilrah, 2023-02-27:
If 125% in OpenTX,through SBUS and into the MSH only results in it seen as 86% in their configuration tool then either there's an option to turn on in it somewhere that you missed or their SBUS implementation is wrong...

ReneHitz, 2023-02-27:
That's exactly what I suspect too, but I don't get an answer from MSH.

kilrah, 2023-02-27:
You never said what receiver / protocol you were using either. Issue could also be there.

ReneHitz, 2023-02-27:
Radiomaster R81, FrSky D8

3djc, 2023-02-27:
Have you tried d16 ? D8 is really very old

ReneHitz, 2023-02-27:
I'll try that. But that won't happen today

ReneHitz, 2023-03-25:
I now have a FrSky RX6R connected via S-Bus. Now I can set the 100%! So I have various S-Bus receivers from different manufacturers that can't do that?!? Learned something the hard way...