opentx/opentx #8926: X-Lite Pro SA missing on display although configured

joerghermann, 2022-05-27:
OpenTX 2.3.15 on X-Lite Pro.
Companion 2.3.15 installed on Mac OS and Windows english.
SA is configured in hardware settings on TX and Compnion as 3 position toggle.

After model loading SA is neither displayed on TX nor Win Companion:

[image]

Side note: The switch order / sequence on this screenshot mirrors exactly the display of the real HW.

Mac Companion 2.3.15 shows the same blank position on the upper right column, but a forth row on the left column, partially overwriting the trim bar. Also note differing switch order:

[image]

**To Reproduce**
1. Start Companion, create radio profile for X-Lite Pro
2. Create new file
3. Click Edit Radio Settings, Hardware tab and check all switches configured correctly
4. Close window, click on Simulate Radio
5. Accept possible errors, navigate to joystick and switch screen
6. Observe white space on top row in right column
7. Mac Companion writes a fourth row on the left switch column, Windows not

On the real TX procceed accordingly.
Right column has white space on top row in right column
Left column does not have fourth row overwriting trim bar.

**Expected behavior**
As per the 2.3.14 screenshot of Win Companion:
All switches displayed
Joystick and switch display mirroring the physical switch layout
Identical display on the TX HW and on Companion across all OS

[image]

**General information**
 - OpenTX version 2.3.15 on Frsky X-Lite Pro
 - OpenTX Companion 2.3.15 on Mac OS and Windows

Zabzab1, 2022-09-23:
I have got same issue. But as a noob need detailed description how it may be solved?
Thanks

o2809013, 2022-09-26:
I have same problem. on 2.3.14 everything works fine. How can i get this version?

Zabzab1, 2022-09-27:
Think need to wait for OTX .16 update. Guess it will be improved....

o2809013, 2022-10-13:
> Trying to solve the problem myself, I assume the problem resides in opentx/radio/src/targets/taranis/board.h line 549. Only 5 switch positions are defined, although X-Lite S and X-Lite Pro have 6 switches. While SF is not explicitly defined, it gets initialized to 0,0, which explains why it draws over SA at position 0,0. My guess is that line 549 should read:
> 
> static const uint8_t switchPosition[][2] = {{0,0}, {1,0}, {0,2}, {1,2}, {0,1}, {1,1}};
> 
> in order to get all six switches shown in the order of their physical placement. Could this please checked or confirmed by someone more knowledgable than me? TIA!

thank you. 

i edit board.h file with this value. now everything matches perfect. 

if anyone interest can follow this guide 

[https://www.rcgroups.com/forums/showthread.php?3824801-Blog-1-Step-by-step-guide-to-building-%28custom%29-OpenTX-firmware-under-Ubuntu-Linux](url)

dolie, 2022-11-19:
Same problem.

dolie, 2022-11-19:
Does anyone can share 2.3.14 bin ?

dolie, 2022-12-05:
> thank you.
>
> i edit board.h file with this value. now everything matches perfect.
>
> if anyone interest can follow this guide [...]

May you open a PR, so it can be fixed on 2.3.16 ?
Or share the bin please ?

o2809013, 2022-12-06:
This is my build. But not work properly. 

[https://drive.google.com/file/d/1tLRiGwfTdMWHSDYri2XxFfnlsjX3i3Zn/view?usp=sharing](url)

I think there is a problem on this step

"cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DINTERNAL_GPS=YES -DCMAKE_BUILD_TYPE=Release ../"

If anyone change this code for X Lite Pro, i can rebuild again.

o2809013, 2022-12-18:
This is my latest EdgeTX build. There is no problem. Everything works. 

[https://drive.google.com/file/d/1Cn5O3Kjxjjy4yR8vprNEtaKzCA9Difcw/view?usp=share_link](url)

first backup your all data. models, sd card, firmware, settings etc.

put this file to sd card --> FIRMWARE folder

on this file, long press enter and flash bootloader.

turn off radio and turn on with holding all trim buttons. 

select flash firmware with this file.