opentx/opentx #8927: Creation and introduction of a subversion system

MlCHEL1234, 2022-05-27:
The bashing, especially against 2.3.XX, does not stop. In principle, it does not have to bother us users. We like using OpenTX 2.3 and we live well with it. We enjoy a quasi real-time service from our devs as far as problem solving and implementation of new features is concerned, which no other system has - as expensive as it may be.

Nevertheless, in various forums and communities, "supposedly very clever people" keep saying that OpenTX is dead etc., we know all the nonsense from some "busybodies", one should switch to Egde or even better to Ethos or Jeti or...

However, this bashing is very bad and possibly even damaging for OpenTX when it comes to newcomers and those switching over. Nowadays, before a decision is made in favour of a certain system, especially for people who are not so deeply involved in the matter, it is enough to have a hint of doubt in order to eliminate this bashed system from the personal selection. Nowadays, no one takes the trouble to find out the background or the truth of such inane statements, the choice of systems is large enough and a system that is talked into doubt is quickly eliminated from one's thoughts. But that does not have to be the case.

I would also like to add that the permanently maintained and continued nightlies with their constant improvements exist somewhat in a parallel-world among users who are not so familiar with the subject matter, completely unjustifiably of course, but nevertheless can be read out again and again in various forum-postings. But it is precisely these nightlies that make up the permanent progress in OpenTX. Elsewhere, problems are not solved, the software remains at a certain level for a long time and it is claimed that one has a mature system... Everyone knows what to make of such funny strategies...

In order to make the lively survival of OpenTX 2.3 clear to the outside world, I therefore propose the creation of a sub-version system. Instead of the nightlies with their numbers, there would then be a consecutive sub-version number, e.g. 2.3.15.1 etc. and no longer 2.3.15 N422 etc. I would think that nothing would change significantly in terms of work for our esteemed devs, only a sub-version number would have to be assigned instead of the nightly number. All that is needed, is an abrupt change in the number system once. From then on, progress in OpenTX would be better recognisable in the same branch as the previous "main system", e.g. 2.3.15, without any further addition. I think this would also clearly enhance the nightlies to the outside world and clearly emphasise their importance as part of the system.

Linux users will immediately know what I mean, as this versioning system has been common in Linux for decades and has thankfully not fallen prey to the inflationary versioning of other operating-systems.

It is clear to me that such a system cannot be set up from one day to the next, but I would nevertheless ask for a thorough examination and introduction, in order on the one hand to shut the mouths of the abusers a little (please excuse my drastic choice of words, but I can't think of an appropriate, softer paraphrase at the moment) and on the other hand perhaps also to make it easier to follow the features.

This Text is translated with deepl.com, as English is not my native language. I of course will answer any questions, especially caused by the translation.

Bye
Michael

lshems, 2022-05-27:
What a nonsense. If naming nightlies would save opentx.

Opentx is mature, virtually bug free, reliable, and useable in preference over lots of other systems for years.

But progress is dead. It is the combustion engine of the opensource radio control systems.

Others have come, running hydrogen, batteries and whatever.

Get real.

kilrah, 2022-05-27:
What would a numbering change achieve?
Nightlies are called that and labeled Nxxx so that it's obvious that they are just untested builds of the code as it stands that night and shouldn't be considered as anything more, that's an industry standard practice.

Trying to "fake" activity makes no sense at all. Sure there's much less than used to be in the past, but that's just how it is. Trying to hide it would be detrimental to everyone.