opentx/opentx #8931: Trim direction is not inverted when input is set to negative weight

HinkePank, 2022-06-06:
Taranis X9D+
OpenTX 2.3.14-otx
2021-07-24 06:08:56

also 

Horus X12S
OpenTX 2.3.15-otx
2022-04-24

Set input weight on elevator to -100
Now elevator changes direction when stick is moved.
Move stick up  => elevator moves down
but
Move trim up => elevator moves up !!!

Expected is of course that trim and stick are working
in the same direction.

To view this go to the servo page on channel 2 (AETR)

kilrah, 2022-06-06:
That's normal. Don't reverse weight in Inputs unless reversing trims is desired.

closed 2022-06-06 by kilrah

HinkePank, 2022-06-06:
Is there any reason for this behaviour ?
I would always expect that stick and trim work in equal direction but maybe I just don't
see ( as a Pilot ) where this can be usefull.

kilrah, 2022-06-06:
Trims, sticks and OpenTX in general can be used with things very different from usual model aircraft control surfaces... ...