opentx/opentx #8932: SD card not detected by Computer

BDL22, 2022-06-07:
Not sure if this is a bug, problem with the radio or the computer.  I have posted information on RC Groups and a number of people have replied that they have the same problem but nobody has come up with a solution.

Radiomaster T16S
Companion 2.3.15. Upgraded from 2.3.14 early May.
Windows 11 with latest updates running on HP laptop. (Windows 10 was upgraded to Windows 11 in December 2021)

Recently, when I connect the radio to the laptop (using the connector on the top of the transmitter) it does not detect the SD card.  It is not possible to write or read models, or download SD card files.

I have tried a new SD card, new lead, uninstalling Companion and reinstalling 2.3.14, but the problem remains.

I have installed companion 2.3.15 on an old laptop which is running Windows 8.  The SD card immediately appears on the laptop and I can read and write to it exactly as it used to work on my new laptop.

Unfortunately I do not know exactly when the problem occurred, but I assume that one of the Windows updates over the last few weeks has caused the problem. 

Is this an issue which needs to be addressed by the team?
Thank you
Brian

3djc, 2022-06-07:
Driver issues on your side. There are rcgroups post that describe how to fix this

BDL22, 2022-06-07:
Do you have a link to the post please as I have not been able to find anything?  Why would the problem suddenly have occurred?  Do Microsoft modify the drivers when they release updates?

kilrah, 2022-06-07:
Displaying card contents requires nothing other than the same built-in Windows driver that any other USB drive uses, and no there's no change to that. 
It's if you try replacing that that you start getting issues...

BDL22, 2022-06-07:
I haven’t tried replacing any of the drivers. I turn on the transmitter with aileron and rudder trims held towards the centre, plug in the usb cable and nothing. No ‘ping’ from the computer and not able to read or write models. Plug the usb lead into an old laptop running windows 8 and ‘Ping’ and read write works fine. It used to work on the new laptop running windows 11 until recently.  !!!

bigtwin, 2022-06-08:
Replacing a driver or messing around with it does not necessarily mean that have to be aware of it. It might be any kind of software which was updated or installed. USB connectivity works for many, many other users and it works with your old computer. So this is obviously not an OTX problem and you are seeking at the wrong place for help on Windows 11.

kilrah, 2022-06-08:
> No ‘ping’ from the computer 

That's at the hardware level, before any drivers are involved. Might be an issue with the computer's port or the cable, a bad connection or such.

BDL22, 2022-06-08:
Thanks for all the suggestions. My computer knowledge is limited.  As it seems to be specifically an OpenTx transmitter and companion problem I don’t know where else to ask for advice.
If I take the SD card out of the transmitter, put it into a usb card reader and plug that into the same USB port on the laptop it ‘pings’ and shows up on the laptop.
If I take my transmitter with the same SD card fitted, same lead, power up with trims held and plug it into my old windows 8 laptop, it pings and I can read/right etc.
When I plug it into this windows 11 laptop, no ping, no access from companion.  The transmitter does say usb connected though.  What I can’t understand is why it was working fine just a few weeks ago.

3djc, 2022-06-08:
Because you deleted/replaced standard driver by something else. It frequently happens when installing drone stuff for example

BDL22, 2022-06-08:
Sorry lack of computer knowledge showing again. So are you saying that if I install something (like drone stuff) this can delete and or replace a windows standard driver?

bigtwin, 2022-06-08:
Yes. You never know exactly and in deep what happens. Installation procedures on Computers asking you this and that and of cause your password. If you dont provide the clicks and infos you dont get what you want. Unhappy we are - we never know what than exactly is done by the installation procedure unless we are in deep skilled specialists.

Also, Windows 11 has changed some stuff in that corner. I know that at least some MPX users also had fights with Win11 and their "USB"-Software in use.