opentx/opentx #8936: 2.3.15 fails to start on X9E

Gfyss, 2022-06-18:
**Describe the bug**
After uploading the firmware for 2.3.15, my X9E fails to start. I can still put it in bootloader mode, and was able to put 2.3.14 back on and boot successfully.

**To Reproduce**
Downloaded firmware file:
  opentx-x9e-horussticks-lua-sqt5font-en-2.3.15-otx.bin

In Companion 2.3.15 select "Write Firmware to Radio" along the left side.

Defaults used, except "Use profile start screen" with custom graphic I've used for years
Check Hardware compatibility is selected.

Click "Write to TX" wait for process to finish and the Close button to become "clickable"

Disconnect USB cable, turn off radio with normal on/off button

Attempts to turn radio on normally result in no response what so ever.  No flashing, beeping, or other indications of life, but can still get to bootloader (thankfully)

Confirmed file written to the ERSKY_9X file system had a current modification date

Attempted above process 3 times, then re-installed the 2.3.14 firmware file with the same name as above (except the last number of course) and radio starts up normally.

Tried 2.3.15 one more time using a different USB port with same results.  Flashed back to 2.3.14 and all is well again.

**Expected behavior**
Should start up as normal

**General information**
 - OpenTX version 2.3.15
 - Hardware FrSky X9E with Horus sticks
 - OpenTX Companion: 2.3.15 for Windows
 - USB3 port with USB2 cable
 - Windows 10 21H2 x64

3djc, 2022-06-18:
I fly X9E with horustick, not an issue. Put firwware on your SD in FIRMWARE directory, start bootloader, flash from there (do not flah with companion)

closed 2022-06-18 by 3djc

Gfyss, 2022-06-18:
So that "worked" but I no longer get the custom screen at startup.  Flashing with Companion has worked just fine up through 2.3.14 and I don't recall anything in the release notes for .15 that say not to use Companion to flash X9E transmitters going forward, or any notice in Companion that X9E updates are no longer supported via that method.  

If this is is just a work-around for a problem that will be addressed in the next release that's fine, but if this is the new permanent solution then Companion needs to be told to stop trying to flash X9E devices and present the other instructions instead.

In summary, while the work-around will get me back in the air, the behavior of Companion and the update process is still not as expected.  IMHO this issue should remain open until it has a more formal resolution documented (why does .14 flash fine, but .15 fail? Is the issue so problematic that the "work-around" needs to be the permanent solution? Update Companion to reflect the final determination)

3djc, 2022-06-18:
It is a somewhat known issue, rarely occurring, and is linked to users pc usb connection and the radio itself. Nothing can be done software side to fix it. Some specific firmware will be an issue for an individual and work great for anyone else, comes next version, others will be affected.

Gfyss, 2022-06-18:
Understood.  I'll keep that work-around in my back pocket.

Thanks for the heads-up.  I'll try a couple of USB connection options to see if any particular port or hub works better.

Gfyss, 2022-06-18:
FYI, I just got it to flash .15 with Companion.  I'm not sure exactly what fixed it, but here's what I did:

- After normal boot, set USB default mode from ASK to USB STORAGE, restart X9E.
- in Windows Device Manager, the TX USB device was reporting "failed to assign address" so I uninstalled that device and re-discovered.  When I did that, Windows requested a volume check which passed with no errors found.
- Unplug USB, restart X9E in bootloader mode, reconnect USB
- Flash .15 firmware with Companion
- X9E now boots to 2.3.15 and shows the custom graphic as expected.

Thanks for the hints on where to look.