opentx/opentx #8939: D8 Mode massiv range issue after upgrade to Opentx 2.3.15

produzilla, 2022-06-26:
After upgrading opentx from 2.3.11 to 2.3.15  on my Taranis X-Lite for beeing able tu use ELRS v2.0 or higher all my D8 Models with integrated D8 SPI Receiver got unflyable: RSSI drops to lower then 30 in a range of 3-4 m with no obstacles in between. Same Issue in D16 Mode with SPI Receiver. D16 Mode with Full-Range Receiver like XM+ seems to work normal..

Rebinding doesnt help/change behavior.
Upgrading internal XJT Model from iXJT_NonEU_20180416 (ACCST v1) to IXJT-XLITE_ACCST_2.1.0_FCC doesnt help.
Creating new Model in Opentx doesnt help.
 
I think it accords to #8421 Add synced pulses to mainstream OTX becuas oher XJT or d8 relatetd issus occurte after this update.

I opend this issue because its not exactly the same like #8846 or #8899 . For my Issue there is no walk around to use d8 mode anymore, 3m to fly is not an option, cant downgrade to opentx because of the need for elrs v2. 

**General information**
 - OpenTX version 2.3.15
 - OpenTX Companion 2.3.15 on Win 64
 - Hardware Frsky Taranis X-Lite
 - integrated XJT Module 
 - integrated SPI D8 Reciver (Happymodel, Betafpv)

Any idea to solve this? If it is really because of #8421 , maybe an option to turn this on/off for explizit models would be a fix?

produzilla, 2022-06-28:
Am i the only one with this issue? Only idea left would be to flash another software like EdgeTx, but i thougt i could avoid this step... downgrading OpenTX  to 2.3.11 would help i guess as d( worked fine with it but then i loose support for two models with integrated SPI ELRS receivers which need opentx v 2.3.12 or higher to work .... really need help or ideas to solve this issue.

Shiva7577, 2022-10-21:
Hello, after update same you, i cant bind d8 models...
You have solution ?

alm-nl, 2023-02-26:
In other cases upgrading the XJT firmware to version 2.1.x solved the issue, I'm using a X9D+ myself with OpenTX 2.3.15 and ACCST v2.1.0 LBT firmware. I have no issues with D8. Kind of strange that it did not work in this case. I would also try with the LBT version of ACCST v2.1.0 and see if that makes a difference for D8 protocol.