opentx/opentx #8941: firmware install

ydyjdwp, 2022-07-03:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

ydyjdwp, 2022-07-03:
Frsky Taranus qx7 running open tx 2.2.0, attempting to upgrade to 2.2.14, when I send to qx7 message is "this version is not compatible with present version on the radio.

closed 2022-07-03 by ydyjdwp

ydyjdwp, 2022-07-03:
I am a total newbe, I am attempting to upgrade my qx7 for use initially with a simulator, sorry if this is invalid, I really want to get this done, and learn something at the same time.  I was attempting the upload through companion. Any help will be greatly appreciated.

3djc, 2022-07-04:
You can seek help on community sites like rcgroups.com or openrcforums.com