opentx/opentx #8948: Trainer mode not working

ramezquitao, 2022-07-25:
I have 2 Horus X10S that I have been trying to use as master and slave  transmitters. In Bluetooth mode, the connection is very unreliable making this unusable. I'm not sure if this is a hardware or software issue as I have not performed any ACCESS upgrade. Today I tried the masted and the slave using a mono and also an stereo 3mm cables, but this does not work at all (not even unreliable).

I'm using Opentx 2.3.15

Thanks

raptor60, 2022-07-27:
I am having the same problems on Qx7s  no trainer function the system recognises the trainer lead  but there is no display of the values of the slave transmitter open tx 2.3.15

CarlovdMeer, 2022-08-14:
I am having the same problems
Trainer mode is not working and telemetry lost messages on Jumper T16 – 2.3.15
Applies:
•	Trainer cable, 3.5 mm jack (mono)
•	MPM module in RX mode (wireless)

In the case of MPM/wireless, I am using ELRS or Crossfire and the Jumper T16 internal module as FrSkyRX. I keep enough distance or use a low power mode on the slave transmitter (MPM) to avoid telemetry lost messages due to high receive strength signals.
I tried DSMRX, but similar issues. 
MPM 1.3.3.7.

**In all cases, the ‘telemetry is lost’ message and trainer mode is not working.**
•	Trainer cable when put in the trainer/DSC plug and trainer mode: Master/Jack.
•	MPM when enabled FrSkyRX and trainer mode Master/Multi. The slave transmitter is OFF, not yet bound or ON/Connected.

It works well on 2.3.14; I used this trainer mode option and wireless MPM last year.

After the downgrade to 2.3.14, everything seems to work as expected.
Applies to the Master transmitter, slave transmitter -> no issues

A minor bug relating to companion trainer mode is a ‘blank field’ in companion 2.3.14 and 2.3.15. Windows 10 and 11. On transmitter device: trainer mode, Master/Multi works; at least a workaround.

On EdgeTX multiple trainer related issues. What about: EdgeTX #2174?

ramezquitao, 2022-08-15:
I made a lot of tests (installing FROS and installing EdgeTX), and found out I was having connection issues. I can confirm the cable trainer works in EdgeTX 2.7.1, and (as OTX and ETX share the same base code) I assume it should work in OpenTX (have not tested and for the moment I will stay in ETX).
The ETX 2174 issue was also mine, and it is now closed as the problem was not with the software but with the hardware (connection problem).
@raptor60 and @CarlovdMeer , have you checked https://github.com/opentx/opentx/issues/6245 ? Maybe your problem is related to this.
I'm closing the issue as the real problem described in this issue was a connection mistake.

closed 2022-08-15 by ramezquitao

CarlovdMeer, 2022-08-15:
I don’t have the time at the moment but can reproduce -at will- with my JumperT16. Maybe I can arrange another Radio (X7, X9D+). They are not unused at the moment.
Then I try the Trainer cable, which fails too.

I usually fly buddy box wireless (MPM, Master/Multi). But, that fails after upgrading 2.3.15; works after downgrading 2.3.14 immediately. 

After downgrading, the Trainer cable works, too (Master/Jack - X9D+ as student radio).

@ramezquitao, your ticket Horus X10S: If it was hardware related, this ticket could be closed, but I will create a new ticket as this is not trainer cable/socket/pin detect related.

ramezquitao, 2022-08-17:
@CarlovdMeer my ticket was indeed hardware related, and the problem was also present when using Edge and FROS. Right now I was able to solve the hardware issue and all is working in Edge with no software change. Next weekend I can install again OTX and see if all works. If you want to reopen this ticket I'm ok.

ramezquitao, 2022-08-20:
@CarlovdMeer I just reinstalled OTX 2.3.15 on my Horus X10S and can confirm the controller works as expected as master and slave in trainer mode using a cable. This confirms my problem was indeed hardware.