opentx/opentx #8949: There is no companion-windows-2.3.16N681.exe at Windows folder download, please check

karmensarodjo, 2022-07-30:
Hi, as subject written, there is no companion-windows-2.3.16N681.exe at Windows folder download, please check...

[image]

thanks in advance!

karmensarodjo, 2022-08-01:
Have Windows nightly builds stopped being supported?

3djc, 2022-08-01:
There hasn't been any changes, hence no new nightlies

karmensarodjo, 2022-08-02:
..oh, but only Windows platform doesn't have 2.3.16N684 version, others (MacOS and Linux) have 2.3.16N684 version. So i thought only Windows version stopped being supported

kilrah, 2022-08-04:
Windows build system is broken, but since there's nothing new no urge to fix. You still got automated builds for linux and mac that are made when there's a commit but since May the only things that have been committed are contributor list changes.

https://github.com/opentx/opentx/commits/2.3