opentx/opentx #8950: Trim travel wrong direction

doloebig, 2022-07-31:
if you set "weight" in Input Tab to negativ value trim travel always positiv

Opentx 2.3.14
TX Taranis X9D plus 2019

3djc, 2022-07-31:
By design

closed 2022-07-31 by 3djc

hrenz, 2022-09-06:
No this is not a design flaw
This is a known bug of openTx
and has been known for many years.

It can be fixed very easily
If the stick is inverted, the corresponding trimmer must also be inverted.

Where was the exact same error
now fixed at ETHOS in the same way

doloebig, 2022-09-06:
i will setup this way 
thx

mpjf01, 2022-09-07:
Seems that hrenz is writing that the same "bug" exists in ETHOS. That's interesting given that we have been told from day 1 that ETHOS is not based on OpenTx and that it's all greenfields design and code (despite the obvious similarities). This is not the first such "bug" that has found its way from OpenTx to ETHOS, the RTC issue is one other.

Eldenroot, 2022-09-07:
EdgeTx should be fine and working, try :)

hrenz, 2022-09-07:
Yes, 
ETHOS had exactly the same error

Together with Bertrand we have fixed it.

If a stick is inverted in the inputs, 
the corresponding trimmer is also inverted.
This solved the problem.

bsongis, 2022-09-14:
@mpjf01 As you are thinking that Ethos is based on OpenTX source code, here is the patch that you will need to fix this issue ;)

[image]

[image]

@hrenz I think it is not fixed yet in Ethos, the PR is open and I have still to push the button, I will do for next release

doloebig, 2022-09-14:
can anyone tell where the code ne put in? i compile my one bin file
I was found many entries on the internet where people report this "bug" but no fix or change for it.
Also i dont know how to change trim setting to negativ.
BTW same Bug in Edge

Reg and THX 
Doro