opentx/opentx #8953: Jumper T Pro not charging. 2.3.15

siogiee, 2022-08-03:
Hi all,

I updated JUmper T Pro to OpenTX 2.3.15 but still can't charging the battery from the remote. When I plug the USB, the power button LED ON, The upper LED ON (only 2 at that time), but it's not charging. So I still have to remove battery and charge it using desktop charger.

I'm wondering if anyone experiencing the same problem and solved it. Thank you in advance.

Cheers!
====

The issue tracker here is for devlopment of OpenTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/) or
[openrcforum](https://www.openrcforums.com/forum/). See also our
[manual](https://opentx.gitbooks.io/manual-for-opentx-2-2/).

For quick questions you can also join our general chat.

Issues that fall into the category of asking for help, will be
generally closed as invalid without an answer.

bigtwin, 2022-08-03:
Why do you think this is an software issue? Related to OTX? 
To me its sounds like Hardware, may Firmware but I would never think of OTX.

3djc, 2022-08-03:
OpenTX or EdgeTx do not have any interaction about charges on those radio

josephduchesne, 2022-09-30:
I've got a Jumper T-Lite and it also doesn't charge over USB. I suspect that there's some sort of hardware bug on the jumper charging board but haven't looked into it. OpenTX has no control over that :/

siogiee, 2022-10-11:
I think this is an software issue because 2-3 of my friend can charge
directly from the remote just after upgrade the software.

But after reading both of the comments, I guess you’re right, it’s a
hardware issue :(