opentx/opentx #8954: 9XRPRO telemetry should be inverted.

jarno-r, 2022-08-14:
**Describe the bug**
9XR Pro with Banggood 4-in-1 module is showing "No Telemetry" as Status. 
This is due to `invertedSerial` not being set. 

**To Reproduce**
Create a model with External RF set to MULTI.
Status will show "No telemetry".

**Expected behavior**
Status should show module firmware version.

**General information**
 - OpenTX version 2.3.15
 - Turnigy 9XR Pro, Banggood 4-in-1 module, firmware version 3.3.14.

**Discussion**
In `radio/src/telemetry/telemetry.cpp:telemetryWakeup()`, if I make the following change, telemetry will work:
````
uint8_t requiredSerialInversion = g_model.moduleData[EXTERNAL_MODULE].invertedSerial;

// Hack to get telemetry working:
requiredSerialInversion=1;
````
How is the `invertedSerial` bit supposed to be set? I cannot find any place in the code where it would be changed. There appears to be code in the Companion for toggling the bit in the module settings, but my 2.3.15 Companion doesn't seem to have that setting.

pfeerick, 2022-09-28:
I'm guessing by now you found out it is a setting buried away in model -> telemetry settings page... but only if you are using a PPM module ;)

https://github.com/opentx/opentx/blob/28f67c1e9a4e949c236f576a215e9becb52fd6ec/radio/src/gui/128x64/model_telemetry.cpp#L69

https://github.com/opentx/opentx/blob/28f67c1e9a4e949c236f576a215e9becb52fd6ec/radio/src/gui/128x64/model_telemetry.cpp#L189-L193

jarno-r, 2022-09-29:
Somehow, I missed that. Perhaps the menu item should be enabled for the multimodule too? Or alternatively, the value should just be inverted for REVX when using the multimodule. Unfortunately, I only have Flysky 2A RXs with telemetry, so I can't test other protocols.

pfeerick, 2022-09-29:
If the MPM needs it generally, I would just invert when `REVX` defined... I guess I could build the firmware and test as I did pull my 9XRPRO out of the cupboard and power it up... :grin: ... was half thinking of driving myself mad with github actions a bit more so I could do a PR on your repo so there are automatic firmware builds for the 9XRPRO ;)

jarno-r, 2022-09-30:
@pfeerick If you (or someone) can test MPM telemetry on the 9XRPRO with some other protocol, that would be great! If it needs to be inverted for `REVX`, I can made a suitable PR for OpenTX. 

If you do have a build script, or can give instructions how to setup an automatic build, I'd be happy to set it up on my repo.
FYI, I've enabled discussions and issues on my repo.