opentx/opentx #8955: Logic switch malfunction with Delta or something I am wrong?

robustini, 2022-08-14:
**Describe the bug**

With ELRS I have the dynamic power activated, and I would like it to play the new power value only if it is held for more than x seconds.
This I managed to do, but unfortunately starting from 10 mW with every small change when it goes back to 10 mW it reads that value.
For example it goes to 50 and back to 10 in less than two seconds, it sounds 10 mW, and with every little variation it becomes boring.
Since the steps are defined (10/25/50/100/250/500/1000) with a=x on each logic switch with a minimum time everything works, but there remains the problem of the minimum power which is read at each variation.
So the solution would be to use Delta, but I've found that it doesn't do what it should.
This thing works if instead of the sensor I put for example the S1 pot.

**To Reproduce**
Steps to reproduce the behavior:

Create a logical switch like this.
[image]

**Expected behavior**
With the logic switch set as in the image it should activate every time the declared sensor has a variation up or down greater than 10 and keeps it for over 1 second, but it doesn't.

**General information**
 - OpenTX version 2.3.15
 - Hardware Horus X10S Express

kilrah, 2022-08-14:
As mentioned by the help text when on the Delay column, "Minimum TRUE duration for the switch to become ON", the condition must be TRUE for the whole duration of the Delay set. 
|d|>x will only be true for a single cycle then false again (unless power changed back and forth every single processing cycle for that one second...) so it simply can't be used for that.

You'll most likely need to write a lua script to do what you're after.

robustini, 2022-08-14:
> You'll most likely need to write a lua script to do what you're after.

Thanks for the reply!
Do you know if there are examples of LUA that can do like this, so that I can adapt them to my need?

kilrah, 2022-08-14:
Nope, don't have a particular starting point that I could think of... a widget that has a background function might be appropriate

Squirlier, 2022-08-18:
Hello,
Try the following.
Keep yout L15  but without any delay or so.
Then add L16 as follows:
Function: a<x    V1: L15    V2: 0       Delay: 1.0
This does the following whenever the L15 detects a change the L16 delay does reset. and only after the delay of L16
and no change happend   ...  L16 get active.

in short: while Sensor(L15) changes Switch is off.(L16)

robustini, 2022-08-18:
> 

Thank you for your suggestion!
If you mean as in the image that I am attaching I have tried but at every change it reproduces the sensor, while I would like it to reproduce it only if that value persists for a certain period of time, as I wrote above.

[image]

Squirlier, 2022-08-18:
OK. So the value should be stable before the Logic Switch activates. IT does this. AS Long AS the value ist changing within the periot of L16 delay.... L16 does Not activate. Maybe you have voice function active and did Not Change to L16.?
Maybe i am getting Something wrong Here.

robustini, 2022-08-18:
> OK. So the value should be stable before the Logic Switch activates. IT does this. AS Long AS the value ist changing within the periot of L16 delay.... L16 does Not activate. Maybe you have voice function active and did Not Change to L16.? Maybe i am getting Something wrong Here.

With the audio set to read L16 reads only 10 mW, ie when for an instant it passes to 100 and returns to 10 stable it reads only that, not the other values if they remain stable for more than 2 seconds.

robustini, 2022-08-18:
No sorry, it works!
But unfortunately it also reads 10 mW, which I wouldn't want it to read, only the other steps.

Squirlier, 2022-08-18:
Good to hear.
Just add another Logic Switch. Which Checks forbgreater than desired value of tpwr. And add IT to L16 as AND so IT does only geht active If the added Logic Switch ist also on (above desired value)

robustini, 2022-08-18:
That's right, now do what I want, great!
I owe you a beer @Squirlier, cool thanks!

[image]

closed 2022-08-18 by robustini