opentx/opentx #8958: Jumper T12 Pro Internal Multi Module "Firmware update error" NoSync

SteveHindmarsh, 2022-08-25:
Hi,
I cannot upgrade the internal multi module (IMM) to https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/download/v1.3.3.14/mm-stm-serial-reta-v1.3.3.14.bin

I am currently running OpenTX 2.3.15 with same bootloader version and these firmware options crossfire, ghost, no heli, internalmulti,multimodule,lua,luac (I don't know why crossfire is listed I don't have that or an other external module?)

I have updated the IMM in the past using the same procedure load firmware from SD [Flash Internal] option, last being V1.3.1.9 RETA however Upg. Advised is now shown since moving upto OpenTX 2.3.15.

I am informed by pascallanger that I have no option but to use a USB serial adapter for use with his Flash Multi tool as the IMM's now also use a bootloader like the external multi module.

Why should this be the case when I have been able to update the IMM in the past, this has effectively ruined perfectly good update functionality, why does it require a new bootloader in the first place?

Can OpenTX not incorporate a fix/workaround for this which does not involve soldering?

Any explanations and help would be appreciated.

Thanks

3djc, 2022-08-25:
There are no known issues with mpm flashing, something must be wrong with your module or radio

SteveHindmarsh, 2022-08-25:
You sound very sure and very dismissive? As stated I have been able to update it in the past and you stating their are no issues seems to fly in the face of what is discussed here: https://www.rcgroups.com/forums/showthread.php?3428887-Jumper-T16-Internal-Multiprotocol-Module-Firmware-Update You should at least have lots of people with early radios who are unable to update the IMM F/W?

3djc, 2022-08-25:
Your link actually proves exactly what I was saying. Those module where defective because the manufacturer did forget to put a bootloader on those module. Nothing OTX can, or should, fix.

SteveHindmarsh, 2022-08-25:
On the contrary, as stated, I have already successfully installed a previous firmware upgrade with no problem proving my unit was not one of those effected. The only thing that has changed on my unit since then is OpenTX.

jtaylor2, 2022-08-25:
You have not said what kind of problem you are having trying to upgrade, only that you can't upgrade.  If you have upgraded your MPM with a lower version of openTX and are really convinced that the 2.3.15 openTX upgrade broke that functionality just backup you transmitter, reinstall the lower version that worked and upgrade the MPM with that version (what version was it that worked).  If that works reinstall 2.3.15 and report back here.

SteveHindmarsh, 2022-08-27:
I have stated in my opening comment what the problem is and what versions were listed so why are you asking me to repeat myself? Anyway, I took your advice and installed a different O/S to see if the internal multi module (IMM) would update as it has in the past from the radio without the need to open up the radio and buy and solder a USB-serial adapter to the IMM as mentioned earlier. The IMM updated fine to v1.3.3.14 using EdgeTx 2.7.1 (same version EdgeTx bootloader too). First time I've tried EdgeTx so I'll stick with that for now but if I revert to OpenTx I'll perhaps try using earlier versions of that to try and find out which version broke the IMM update process, Thanks for your help.

jtaylor2, 2022-08-27:
> I have stated in my opening comment what the problem is and what versions were listed so why are you asking me to repeat myself? 

Sorry I missed the "no sync" in the title, was looking for the problem details in the comment.  And I still don't see (unless I missed it too) what openTX version worked when you were able to flash it before.  I had and fixed a similar problem a couple of years ago with flashing a avr MPM module ttps://github.com/opentx/opentx/issues/8060 but I don't have a T12 so can't test your problem.

In any case glad you got it updated and if you find out what version broke it post it here.