opentx/opentx #8961: Wireless trainer with V1.3.3.15 not working on TX16s and DX9

Dave-thirteen, 2022-09-02:
I have a small problem with wireless trainer and opentx 2.3.15

My setup is TX16s with Jumper 4 in 1 module added to the back. This is the master TX.

Both RF units are running mpm  V 1.3.3.14 in TAER format.

The student TX is a DX9 in DSMX mode.

I had great trouble in binding the two together and could only get it to work randomly and I could not document a procedure that would cause it to happen, or not happen.

After I got the two TX's to bind I could not fly the model - see the video.

When trainer is selected the control surfaces on the plane randomly jump about.
I can see the problem when using the output monitor function on the TX16s.
I also see the problem showing up when viewing either the internal  or external RF modules status on the bottom of the model setup screen.
Both  modes keep jumping about and I attempted to capture this on the video with the last two images showing the working state and then the problem state. You can see the end of the mode line changing!

I may have got round the problem by changing the TX16s firmware to version 2.3.14.
This appears to work as I expected with one very slight problem (that may to do with the way it works anyway). This shows up as the control surfaces appearing to have a lower resolution when in trainer mode rather than in master mode.

I'm not sure if the problem has been solved so ask for help from those in the know!

Thanks in advance.

Dave.
https://youtu.be/7SY7pPtLo3g

ThePhantom05, 2022-09-30:
I support it, it does not work on taranis X9D +. From the word at all. not where.

3djc, 2022-09-30:
> I support it, it does not work on taranis X9D +. From the word at all. not where.

It is definitely not working on any frsky radio, this is due to technical choices made by frsky

ThePhantom05, 2022-09-30:
Won't OpenTX fix it?

3djc, 2022-09-30:
Multi trainer mentioned in this issue never worked, and never will on frsky radio, due to frsky hardware decisions

ThePhantom05, 2022-09-30:
Not true) on the firmware it seems to be 2.3. 12, if my memory does not change! Everything worked, I used - Fly Sky 2 ch - 3.5mm wire and everything worked - I taught the student even the settings remained. And in version 2.3.15, there is nothing except the screen with the inscription slave.

3djc, 2022-09-30:
I'm sorry but you are wrong, it is even clearly mentioned in multiprotocol documentation : https://www.multi-module.org/using-the-module/wireless-trainer

ThePhantom05, 2022-09-30:
Tomorrow! I will write to a friend who has not been updated and ask him to throw me the screen of his remote control (with a working coach. Although there is a little evidence. Here https://alnado.ru/forum/viewtopic.php?p=2550&hilit=%D1%82%D1%80%D0%B5%D0%BD%D0%B5%D1%80+%D1%83%D1%87%D0%B5%D0%BD%D0%B8%D0%BA# p2550 . Link to the forum... It's not possible to configure it right now.

ThePhantom05, 2022-09-30:
So don't say I said it wrong....

3djc, 2022-09-30:
Oh Russian ? No help from me then, sorry

ThePhantom05, 2022-09-30:
What is the problem?

ThePhantom05, 2022-09-30:
> О, русский? Тогда никакой помощи от меня, извините
What is the problem?

ThePhantom05, 2022-09-30:
> О, русский? Тогда никакой помощи от меня, извините

But I'm not sorry! What then?

kilrah, 2022-09-30:
> Link to the forum... 

As far as I can guess this is wired trainer. The discussion is from 2018, AFAIK multi wireless trainer which is the subject of this issue didn't even exist then.

ThePhantom05, 2022-10-01:
> > Link to the forum...
> 
> As far as I can guess this is wired trainer. The discussion is from 2018, AFAIK multi wireless trainer which is the subject of this issue didn't even exist then.