opentx/opentx #8962: Not able to download the companion from opentx

Snoo1739, 2022-09-03:
Not able to download opentx 2.1.9 from the website as it's giving a 404 error

0815Creeper, 2022-09-03:
All but the 2.3 release are missing: https://downloads.open-tx.org/

Snoo1739, 2022-09-03:
But I have the 9x stock board and the only compatible version with it is 2.1.9, 2.3.x are of no use to me right now
Is it permanently deleted or will it be available later?

0815Creeper, 2022-09-03:
that is exactly what I want to know too ;)

bigtwin, 2022-09-03:
As far as I know (but I am pretty sure about that) you can only download the very latest and current OTX version. OTX is compiled on demand, just according your "_Build Options_" and than copied to your Computer.
You may have success if you find someone who still has the version 2.1.9 as you need it available. But you dont get it the "official" way.
I had this once, when I was looking just to one earlier patchlevel of the 2.3 stream. In the URL above you will only find the latest version (currently 2.3.15).

Snoo1739, 2022-09-03:
I tried with the companion9x version too, but its downloading a file of size 185 bytes which is not compatible nor writing on the board, what do you suggest should I do other than finding someone with the otx..

0815Creeper, 2022-09-03:
> As far as I know (but I am pretty sure about that) you can only download the very latest and current OTX version. OTX is compiled on demand, just according your "_Build Options_" and than copied to [...]

You are wrong. The major versions (2.0.x 2.1.x and 2.2.x) should still be available, only the subversions are no longer available as Transmitter firmware. Unless the devs recently decided to no longer provide old major versions, but that would be news to me.

0815Creeper, 2022-09-05:
please reopen, it is not fixed!

bsongis, 2022-09-05:
We have had to move on a new server!

closed 2022-09-05 by bsongis

0815Creeper, 2022-09-16:
Ok, but still not fixed ;)

0815Creeper, 2022-09-16:
or are they lost forever?

jtaylor2, 2022-09-16:
What are you trying to download and what are you using to download it?  I can download  Windows Companion 2.1.9 from http://downloads-21.open-tx.org/companion/companion-windows-2.1.9.exe

If you are trying to download it by clicking  the link on the page at http://www.open-tx.org/ using Chrome or Edge your browser is probably blocking it because it is a http link and you are clicking on it from a https page.  Either use a browser that will do what you want, instead of what it wants you to do or  right click on the link, select "copy link location" and past it into the url bar of another window or tab.

Jim

0815Creeper, 2022-09-16:
All but the 2.1.9 and 2.3 release are still missing: https://downloads.open-tx.org/

Offical links therefor not working. Eg.
https://www.open-tx.org/2019/07/11/opentx-2.2.4

nostromo123, 2023-01-12:
2.2.4 is online again!!

br KH