opentx/opentx #8963: Taranis Q X7 Access keeps draining USB-plugged smartphone battery (or other power sources)

necarnot, 2022-09-05:
With a (successfully upgraded) 2.3.15 OpenTX firmware, we're connecting this Taranis Q X7 Access radio to an Android smartphone with an USB cable, and we're playing FPV FreeRider Recharged with no issue.

But the problem lies in that the radio is rapidly draining the smartphone's battery (rapidly like in "a couple of minutes, and alas efficiently").

The battery in the radio is Gens Ace LiPo 3800mAh 7.4V (2 cells) which was already fully charged.

We're witnessing the same behavior when plugging it to a laptop.

My question : is there some setting to prevent the radio from draining power from the USB connected devices?

geofrancis, 2022-09-05:
your phone is in OTG mode so its trying to power the taranis, you need a dock with power delivery pass-through so it can be charged and be usb host at the same time.

3djc, 2022-09-05:
Nothing otx related, or that otx can help with

closed 2022-09-05 by 3djc