opentx/opentx #8965: Trainer follows Models rather than System.

sande005, 2022-09-12:
System/Trainer works well (wired connection), but requires reconfiguration when a different buddy box is used.
For those of us with multiple students bringing their own transmitters, it would be great to have the Trainer config be custom to the Model rather than the System.  Hence we could have "Apprentice - John" vs "Apprentice - Fred" as models for the same plane, but different buddy boxes.

leslieadams57, 2022-09-12:
Dunno if it helps but you can already do this. Instead of using a standard trainer connection use channels (TR1,TR2,TR3,TR4 etc. Trainer channel 1,2,3,4) as switched inputs. This way you can assign any trainer channel to any input device. I also use this to assign right stick only (mode 2) to the student. See Below ...
[image]

sande005, 2022-09-12:
leslieadams57 - That resolves channel order, but not differences in weights (Spektrum vs OpenTX etc.) or centering.

Perhaps even better would be instituting a few (2-4) "Buddy Box Profiles" based on the current Trainer setup module, that can be preconfigured and then selected when training multiple students, each with their own buddy box.  Then only one Model is needed.

leslieadams57, 2022-09-14:
> leslieadams57 - That resolves channel order, but not differences in weights (Spektrum vs OpenTX etc.) or centering.
There's no reason that you couldn't have multiple definition of TRx's with different weights switched via a "profile" switch. You would still only need one model definition. E.g. Switch "X" up is FrSky, middle is Spektrum and down is Futaba. The options are endless.

One other thing. Most of the guys that I train tend to have their own aircraft/Tx. Normally it's this Tx that becomes the "master" and specifically defined for their usage, with trainer/buddy added for training. Multiple definitions in this circumstance wouldn't be valid.

Either way would work. I've found that what's there already in OpenTx will do practically anything. Someone somewhere has a way of doing virtually anything.

sande005, 2022-09-14:
leslieadams57 - true, using OpenTX as the buddy box to a SOB (Some Other Brand) TX has it's own challenges.  Beyond the scope of this suggestion.

To illustrate the problem with using TRx channels -
Spektrum output is 78% of base OpenTX, so an OpenTX Model using 100% Inputs/Outputs would need 128% replacement Input from the Spektrum Buddy Box to have the same throws.  OTX Inputs can't go above 100, so one has to use Outputs to scale.  This can have issues with correct functioning of rates, expo, curves and mixes since they are seeing reduced Input values and the scaling doesn't happen until the end. 
The current System/Trainer module allows scaling (and order, centering) to happen before it is used as an Input source.  So no changes to the buddy box  OR the OpenTX Model otherwise needed.  Hence why having a few "profiles" of different buddy boxes would be handy in situations where there are multiple students being trained on the same model.

sande005, 2022-09-14:
For further thought/discussion - having the buddy box be part of the model, and therefore requiring different copies of the model for different buddy boxes may reduce the the issue of incorrectly picking the wrong profile.  Or, perhaps the trainer profile name could be appended to the aircraft on screen name.  Or a warning box at startup to notify which trainer profile is active....