opentx/opentx #8967: tempereture sensor limited to cont of two

khmewes, 2022-09-12:
Hi, I am trying to add more than two temperature sensors (4 in total) to my 4in1 Devo protocol telemetry using an OpenTX Radiomaster TX16S 4in1. So far no luck it will only discover two sensors, T1 and T2. I am sure I have 4 sensors T1-T4 using the Walkera CTL01-D telemetry module with a MX400 quad. It works with the original Walkera Devo F12E and with the Deviation Jumper T8SG 4in1. My question is, does OpenTX have a limitation of two temperature sensors and is there a way to get around this, i.e.add more temp sensors?
Best Regards

3djc, 2022-09-12:
Otx is not limited sensor wise (well it is by the total amount of sensors). Could be linked to the multi and its implementation of Devo telemetry

khmewes, 2022-09-12:
Thanks. The Devo implementation maybe the reason. I have to follow up in the Multi/4in1 forums on that. Just curious, what is the total number of sensors available in OpenTX? And are any of the other seosor types, like GPS, RPM, etc. limited in number in their groups themselves? Like how many RPM sensors for instance.
Thanks again for the info and Best Regards

3djc, 2022-09-12:
Your max sensor varies by radio type (because of memory constraints) but is set at 60 for Horus for example. There is no max number per type, you can have 60 temp sensors if you want

khmewes, 2022-09-12:
So the limitation will probably be in the detection algorithm, i.e. what the firmware is programmed to look for and recognize,  like sensor addresses (0-FFFF hex for instance). So it seems in my case the  additional temp sensors (and addresses) that the telemetry module is communicating are not 'discovered' and recognized in the Devo 4in1 implementation. And it is not a limitation in OpenTX.

khmewes, 2022-09-12:
Interestingly in the Devo 4in1 implementation on the Deviation-based Jumper TSG8 radio the 4 temp sensors are discovered. So the 'Multi' people that 'ported' (perhaps?) this protocol to otx did not completely/correctly do so. I would like to contact them and ask for a fix if possible. Do you have any contact info for them?
That would be great.