opentx/opentx #8969: "Unknown server failure, try later" error when trying to compile FW on 2.1.9

Gloopann, 2022-09-19:
Been trying to flash my old Turnigy 9x with OpenTX to breathe some new life into it, but I can't seem to download the firmware onto my PC. 
I get the below attached pop-up window, and the program creates a HEX file that simply says **"ERROR: 125"** when opened.

[image]

I replaced the old ssl libraries with the new ones, that didn't seem to fix anything.

pfeerick, 2022-09-26:
Most likely either it isn't as simple as just replacing the DLLs, or it's because you can't download older firmware than the current release (2.3.15 at the time of writing).

3djc, 2022-09-26:
We had to move server to another, cheaper, hosting, unsure if 2.1 compiles where transfered

RybOlya, 2023-01-02:
@3djc Hi there, encountered the same problem, but what is the solution if your server moved and I can`t download the specific 2.1.9 v, needed for the Turnigy 9X. Will be thankful for suggestions

gtagomez1, 2023-01-03:
I have the same problem, has anyone found a solution? or does anyone have the .hex for the Turnigy 9X?

pfeerick, 2023-01-03:
I actually do still have a 2.1.9 fw knocking around... hope it helps. 
[opentx-9x-v2.1.9.zip](https://github.com/opentx/opentx/files/10335691/opentx-9x-v2.1.9.zip)

tomekmgr, 2023-02-01:
Same here for me, need 2.1.9 for 9xr, m128 with few features.... Trying to compile by my own but error after an error....

MadMark13, 2023-04-10:
Same again for a 9XR. any updates?  Not sure I'even up for compiling.