opentx/opentx #8971: 2.3.16 no milestones and night version

ThePhantom05, 2022-09-29:
Hello, dear developers) Where did the night version 2.3.16 go and why is it not in Milestones! What happened?
P. S. And for a long time)

shimsquat, 2023-11-03:
Also desperately looking for a 2.3.16 nightly for support of the new Archer Plus receivers.

raphaelcoeffic, 2023-11-05:
> Also desperately looking for a 2.3.16 nightly for support of the new Archer Plus receivers.

Look, this is the beauty of open source. You have a few options available:
- compile yourself, not that hard
- ask a friend to do it for you (might work)
- wait for some more
- switch to EdgeTX (the new receivers should be supported in coming `v2.9.2`)

maschibl, 2024-02-28:
Is there a very rough ETA of 2.3.16? Weeks, Months, Never? I would appreciate it for the same reason ;-)

pr-newman, 2024-04-23:
There is this: https://github.com/opentx/opentx/issues/9015
and there is this: https://openrcforums.com/forum/viewtopic.php?p=157368#p157368

It is hard to join the dots "officially"?  Any opinions from inside or outside FrSky appreciated if we really are being abandoned. 

And by the way, FrSky and other manufacturers, I bought your hardware in order to get OpenTx, not vice versa.  If you remove official support for OpenTx, I am likely to stop buying your hardware entirely and start buying from a manufacturer who does.  Don't mistake one loyalty for another.  ... But I digress.  --Pete

gagarinlg, 2024-04-23:
> And by the way, FrSky and other manufacturers, I bought your hardware in order to get OpenTx, not vice versa. If you remove official support for OpenTx, I am likely to stop buying your hardware entirely and start buying from a manufacturer who does. Don't mistake one loyalty for the other. ... But I digress. --Pete ps I donate to this project, too.

all other manufacturers moved to EdgeTX because of the lack of progess on OpenTX