opentx/opentx #8974: Opentx 2.3.15 HorusX10s Express Telemetry Lost

nightfury24, 2022-10-05:
Since I have updated from 2.3.14 to 2.3.15. I have very often telemetry lost and recovered + randomly there is a significant delay of telemetry update on screen ( if telemetry recovered) + takes approx. 5-10 sec to get a connection established between receiver and TX ( not very often, 1 out 10 times ) after powering up the receiver.

Never had this issue with 2.3.14 ( been working since the beginning of the version came out) 

I lost my model recently, the connection was lost in mid-air.

More embracing is that OPEN TX does not allow to degrade the firmware. It should be allowed to degrade for at least 3 months after launching the new one? I have to find my mate if they one in the backup, till then won't be flying or if can't find probably will update to ETHOS.

**General information**
 - OpenTX version 2.3.15
 - Hardware: Horus X10s Express
 - OpenTX Companion: 2.3.15

builda1, 2022-12-02:
I always keep old versions of companion on my laptop in case i have to rollback to an old version (as i did with 2.3.15 on same radio for same reasons-i went back to 2.3.14 and the problem went away. Look on your computer for the old installer for the last version of companion, uninstall the current one, then install the older one. From here you will be able to download the older firmware and write it to the radio. I learned this trick many years ago when I discovered you could only go forward not back-so I started hanging on to old install files for companion. I've actually ended up putting ethos back on my X10SE because it is 1000 times easier than OTX plus I'm also using a Tandem X20S so it did not make sense to have one on OTX.

3djc, 2022-12-02:
There is absolutely no point in keeping companion version, and no, an old version does NOT download an older firmware, only the latest version can be downloaded. If you are downloading a firmware today with companion 2.3.1, 2.3.14 or 2.3.15, you get the same firmware, 2.3.15 as of today

builda1, 2022-12-02:
> There is absolutely no point in keeping companion version, and no, an old version does NOT download an older firmware, only the latest version can be downloaded. If you are downloading a firmware today with companion 2.3.1, 2.3.14 or 2.3.15, you get the same firmware, 2.3.15 as of today 

I guess I must have kept my original download files as well, because I was able to reinstall the old version , I've done it for others as well.