opentx/opentx #8975: Si switch in tranis 2019, missing pot and multi position selections, this woud be a great place for a 6 position switchrepalcing the unusful push button

tj51, 2022-10-10:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

3djc, 2022-10-10:
Hardware wise, switches are connected logical inputs, not analog, cannot become pots (analog)

closed 2022-10-10 by 3djc

tj51, 2022-10-10:
Thank you, I guess the only logical choice is s2, Tj