opentx/opentx #8976: Download Link for all 2.2 Versions not working

SpaceToaster2, 2022-10-15:
Hi Guys, 

it is not pissible to download any 2.2 Version. 

Link is not working "https://downloads.open-tx.org/2.2/" 

is there a reason? 

Best regards

MlCHEL1234, 2022-10-15:
Hi,

at

https://downloads.open-tx.org/

you can see, what you can download...

3djc, 2022-10-15:
2.2 are i think not available anymore for downloads

3djc, 2023-01-10:
Not firmware for 2.1, so mostly irrelevant

nostromo123, 2023-01-10:
But irrelevant 2.1 companion can still be downloaded.

[image]

People here are asking not for companion 2.1, but for 2.2, which for some here still seems to be relevant somehow.

In my case, I have two Taranis X9D here, happy on 2.2.4 and a new Linux Debian version is on the horizon.  So I am looking, what of my software is working on that new version. Therefore I was looking for the companion22_2.2.4_amd64.deb package. I just downloaded companion21_2.1.9_amd64.deb, but that is for sure not helpful.

And I think, I am not alone with that situation.

br KH

0815Creeper, 2023-01-10:
duplicate to #8962

developers do not seem to care...

nostromo123, 2023-01-10:
I realize here, that I am not alone with my request, and i guess , that only around ~ 1 out of 10 OTX users affected will report here in github on that missing software issues.

br KH

nostromo123, 2023-01-12:
2.2.4 is online again!

br KH