opentx/opentx #8980: Firmware 2.3.15

joblackk4949, 2022-10-22:
Just flashed my radiomaster as i needed the heli menu.   This thing is crap.  I had previously good success with frsky receivers.  No issues for years.   All receivers now are failing, even  the vr once.   They bind but intermittently they lose connection, and servos are all going into hold.   Terrible version.  I have to downgrade.

bigtwin, 2022-10-22:
This issue and your contribution is senseless and not helpful.
2.3.15 is available since many month and many people are running it with success. Just this fact shows you that your problems are not related to 2.3.15.
If you would describe your problems with some technical info rather than grumble around you may receive help to solve your problems. But I feel like you have some simple usage problems. In order to get help here, you are doing better to consult e.g.  rcgroups.com

joblackk4949, 2022-10-22:
Thanks for your post although it is quite condescending.  My advice,  if you don't have anything of value to add, don't post. My post is quite simple.  I have no problem binding any of the receivers.  They do bind, but all and I mean all frsky, spectrum, even the radiomaster R86,  while bound and moving the controls,  every so often the wireless connection will start blinking telling i lost signal (including radio voice).    The reason i am posting here and not in rcgroups is because,  while you think few months is enough to be made public  ("many people" is subjective unless you know exact how many have used this version),  I fear this version has not been tested well enough to be made public.  Some (including me) may use this on helicopters.  The outcome would be devastating to find such problems in flight where, for no apparent reason we lose signal and model.   I upgraded from 2.3.9 with absolutely no issues before.   After the upgrade, radio no longer works as expected.   You don't think this is related to the 2.3.15?   Enlighten me.

3djc, 2022-10-22:
Many people as over 100k...

joblackk4949, 2022-10-22:
Be that as it may,  my problem still persists.  Previous person suggested that i am doing something wrong and that my problem is not related to the upgrade.  Trying to understand what i did or i could do wrong other than bind.   Is calibration required after an upgrade?  Thanks.

MlCHEL1234, 2022-10-22:
Hi,
where in the OpenTx-Manual does the result differ from your experience? Surely you have read up on how to proceed in the manual?

joblackk4949, 2022-10-22:
My expectation/experience is that upon binding a receiver,  i expect it to stay on and only lose connection outside of the range but not 2 feet from it.   I am experiencing general connection issues related to the new firmware.  This is does not constitute a personal problem as you bluntly suggest.  Are you saying somehow my binding process is wrong?    Please read the OP question.  I fear you didn't even glanced over it but attaching once credibility when you don't have an answer.   Don't need a manual to bind a receiver.  With 20 years RC experience I don't need a manual for that to know when a receiver doesn't communicate with the radio correctly when all have that idiot proof flashing LEDs. 

Please move along if you feel the need to troll.   I asked a valid question that all of us go through it every time we setup or fly (i.e. binding the radio to a model), yet i get slammed for somehow being ignorant.  There aren't many options here to look for, which is why i am suggesting perhaps this firmware is at fault: either it works or it doesn't.

3djc, 2022-10-22:
The thing is Otx doesn't deal at all with RF link. If you are suggesting your range is really short, then it rules out Otx

joblackk4949, 2022-10-22:
Where in this entire post did I mention RF link man?  I have a set of pre-bound receivers with no issues prior to the upgrading.  Now all are not working and what I am getting is every so often the connection drop with the following messages:  rfsignal loss, or rfsignal low,   telemetry lost, telemetry recover.. etc.   Once again,  I am able to bind them.  They just lose connection (as in wifi connection loss).

joblackk4949, 2022-10-22:
https://user-images.githubusercontent.com/25379247/197358646-6f52d9ee-47e0-4fb3-ab16-5e5f55bf53b5.mov

jrwieland, 2022-10-22:
I fly helicopter exclusively. I am trying to assist with yiur issue not troll.
What version did you upgrade from?
Your initial post mentioned you were going to downgrade.

I see you're using the radiomaster tx16. So do I.  This radio transmits using the  internal 4 in 1 module.   Did you also upgrade the firmware of the module?
If  the previos firmware was quite  outdated this could be a cause for the issue.

Additionally,
 Have you gone thru the process of fine tuning the receiver after upgrade(s).

I saw your video and it is browning out with excessive servo movements.   After the brown out it reconnects and your able to move them again. I had this issue with a radiomaster receiver.  The receiver requires 5 volts to operate with the full load of all of the servos & telemetry,  the remaining voltage for the receiver was insufficient intact it  dropped to 3.6 at one point this caused a brown out.  The Gyro (BeastX) reinitated and naturally a crash occurred.  After reviewing the logs after thr crash I saw what happened.  A larger bec was installed doe to the load being used by the hv servos.

joblackk4949, 2022-10-22:
Thanks John,  much appreciated. 

I've mentioned the downgrade if i can't resolve my existing problem.   I've upgraded from 2.3.9 for the simple fact to add the heli menu,  to 2.3.15 (figure it wouldn't hurt).    I was quite contempt with the way the radio functioned before but only flown fixed wings and mostly gliders (dlg in particular).  Starting from the back:  Resurrecting a trex that hasn't flown in some times,  power for sure is adequate,  30AESA with 1500 3S pack.   Flown this before with no issues.   After the Taranis debacle i switches to radiomaster and i was pleasantly surprised to be able to bind just about anything.   I've been flying several F5J gliders with this particular radio with frsky modules with zero issues until now.   Now,  trying to understand when you are saying "module" are you referring to the receiver?   With Taranis there was and i've done it several times, flash a receiver firmware through the back (behind battery case) port.  My frsky receiver have been flashed about a year ago so doubt they need anything.  However,  as mentioned above i have 5 different type of receiver used prior (spectum, frsky X modules severa, frskyk G with vario and telemetry, frsky V models, flysky,  radiomaster F86 receiver, and even a dsm cheap receiver used for testing only.   All and I mean all are doing the exact same thing you see in the picture.   I would switch radios before i venture flashing all receivers. 

Thanks again.

jrwieland, 2022-10-22:
No I'm not referring to the receiver the actual 4 in 1 module has its own firmware separate from open TX

Open TX passes the inputs or rather outputs through this module and then it (4 in 1) actually transmits to the receiver

Here are the links for the multi module so you can get the latest firmware it may solve your issue

https://downloads.multi-module.org/

And here is the page to his github
https://github.com/pascallanger

joblackk4949, 2022-10-22:
Thanks,  
I did read about the multi module however many agree there are not needed unless there is a good reason when new hardware fails compatibility test.  In this case there isn't any and backwards compatibility is and should always be kept.  I will look into it and perhaps do the module upgrade.  I am tore as now i don't trust thing thing.  I might very well go back to version 2.3.9.

jrwieland, 2022-10-22:
I'm afraid you're mistaken on this one...

 openTXVersion 2.3.10
Recommended an update to the multimodule to a minimum of 1.3.1.69

Open TX 2.3.11 update Incorporated changes made to the multi-module no update was required /Recommended though

Open TX 2.3.12 recommended update  to 1.3.2.84

Open TX 2.3.13 ,14 and 15
Do not recommend an update due to changes in either firmware

joblackk4949, 2022-10-22:
Ok, thanks. 
I'll do an update and report.  

Once again, much appreciated for this valuable info.

joblackk4949, 2022-10-23:
Just finish the update.  It did not help.  attaching video.  First few few min, showing current version 2.3.15 than going to the model module showing 1.3.3.20 firmware of the module.    Still getting drops.

joblackk4949, 2022-10-23:
https://user-images.githubusercontent.com/25379247/197397252-9c7b9259-1759-455c-ab7d-e8f80d495d59.mov

Guarana83, 2022-10-25:
Hello. @joblackk4949. When you test your receivers at home, move the remote control away from the receiver under test and if possible switch to "low power" mode. Have you thought about doing the frequency adjustment?
Have you tested with batteries instead of BEC(UBEC)?