opentx/opentx #8983: macOS Ventura, no radio found

christophe-calmejane, 2022-11-22:
**Describe the bug**
My radios are not visible to Companion, although they are to Finder.
I tried manually listing DFU devices using dfu-util from the app, nothing found.
When first plugging the radio to USB, Finder asks for permission to read from it, so I suspect Companion is not properly asking for permission to read from USB.
I'm using a M1 macbook with macOS Ventura.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug a radio (in DFU) to a macBook using Ventura
2. See it (and the SD card) in Finder
3. Try to synchronize using Companion
4. Get "no radio found" error

**Expected behavior**
Able to interact with the radio from Companion

**General information**
 - Hardware tried: Frsky X9D+, X7S
 - 2 different cables, including the one provided with radios, also tried directly to usb-c (with usb-a -> usb-c adaptor) and using a usb-a hub
 - OpenTX Companion 2.3.15
 - macOS Ventura on a M1 macbook pro

jtaylor2, 2022-11-23:
Perhaps same as this:
https://openrcforums.com/forum/viewtopic.php?f=94&t=14030&p=156830#p156830

christophe-calmejane, 2022-11-24:
I confirm the fix works, I'll propose a PR, thanks a lot for the search (it was posted after I created the ticket :D)

3djc, 2022-12-20:
Fixed by https://github.com/opentx/opentx/pull/8984

closed 2022-12-20 by 3djc