opentx/opentx #8992: OpenTx not playing 'timovr1.wav' (or 2 or 3. Radio is Taranis X9D+ 2019, OpenTx version is 2.3.14-otx (1e09791a), dated 2021-07-25.  SD card contents is for installed version of OpenTx & the correct 'opentx.sdcard.version' file in the root of the SD card appears correct (2.3V0038).  When using timer 1 or 2, when timer expires, no audio file is played. Countdown (beeps, haptic, voice) all work OK)

Marvin-of-Mars, 2022-12-27:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

jrwieland, 2022-12-27:
The sound files are limited to 6 letters.  Any more and the system does not recognize them.   Shorten the name since they are all 7 letters and they will work.

Marvin-of-Mars, 2023-01-13:
Thank you for your reply, however, I believe that you're wrong regarding filenames.

I'll explain:

In the system sounds folder/directory, there are multiple files with filenames longer than 7 characters & they work OK. 
E.g. 'rssi_red.wav' (clearly 7 characters) and spoken text is 'RSSI critical' & that works OK.

Additionally, the sound pack in use came from the OpenTx Web site & is the correct version for the version on the Tx (OpenTx 2.3.14). Further, there are three timers available and consequently, three 'timover' files - 'timover1', 'timover2' & 'timover3' - seems to make pretty good sense.

My initial thought was that the files were corrupt, hence the reason for sourcing and installing the correct sound pack from the OpenTx Web site.

This issue has been tested on another Taranis X9D plus 2019 and the result is the same.

Also, I have an earlier Taranis X9D plus (i.e. - not a 2019 version) running OpenTx 2.2.1 & this works OK with 7 character filenames - i.e. at the expiration of timers 1, 2 & 3, the appropriate announcement is played.

Further suggestions/comment is invited!

jrwieland, 2023-01-13:
you are correct some of the system sound files do contain more than 6 letters.  I am assuming that you are creating custom sounds in lieu of the standard sound pack.  Normally the wav format is off when creating them.  I have found this to work every time for me (all of my sounds are custom)

They must be converted to mono, preferably normalized, and encoded in Microsoft WAV 16-bits signed PCM at a 32 kHz sampling rate, you can use [Audacity](http://www.audacityteam.org/) to do that, it works great. (I have had troubles when the sample rate is 16 or 8) as stated in the manual

Here is the manual for reference
https://doc.open-tx.org/manual-for-opentx-2-2/advanced-features/audio

I Hope this helps, If not I could take a look at the files and see if I can get them to work.