opentx/opentx #8993: RadioMaster TX12 MkII connection problem

hdlbq, 2022-12-31:
Hi,
I start my TX12 in Bootloader mode 2.8.0.
I connect the USB cable to the TX12 and the PC, and the TX12 displays :USB Connected.
OpenTX Companion displays the "No Radio Found alert" as soon as I try any of the actions of the Read/Write Menu:
- Read Models and Settings From radio
- Backup radio to File
- Read Firmware from radio
I'm pretty sure that the cable is a "good" one, since I use it to get the TX12 contents while doing:
I started the TX12 in a normal mode,
I connected the USB cable to the TX12 and the PC,
I choose "USB Storage mode"
I can browse the SD Card and TX12 contents with the Windows Explorer.
 
Software Versions:
 - OpenTX Companion 2.3.15 on Windows 10
 - TX12 Bootloader 2.8.0
Thanks for your help.

rotorman, 2022-12-31:
Your TX12 mkII likely was flashed with EdgeTX and not OpenTX from the factory (assuming from v2.8.0 version string of your Radio Firmware bootloader).

Thus you might want to use EdgeTX Companion instead? The latter is downloadable under Assets of the following link: https://github.com/EdgeTX/edgetx/releases/tag/v2.8.0
Here also link to EdgeTX Discord, in case you have further questions: https://discord.gg/wF9wUKnZ6H

hdlbq, 2022-12-31:
Hi,
thanks.
I will try it.

hdlbq, 2022-12-31:
Actually, You're rigth, and I apologize for wasting your time.
Thanks

rotorman, 2022-12-31:
No problem, glad to help. Thus the issue here can be closed?

hdlbq, 2022-12-31:
Yes.

closed 2022-12-31 by 3djc