opentx/opentx #8995: carrytrim as Boolean, wrong in LUA function "model.getInput" and "model.insertInput"

ChrisOhara57, 2023-01-03:
Hello
I do have a problem with the LUA function "model.getInput(input, line)" and "model.insertInput(input, line, value)".
As per Reference manual there is a parameter "carryTrim (boolean) input trims applied".
This definition seems to be wrong cause it always returns "Zero".
Also in OTX 2.3.15 the option menue (for trims in Input lines) shows different possible values and not a Boolean.
Values for Trim shown are "off, on, RUD, AIL, ELE,TRO".
(same for Reference manual of edgeTX and it's also beeing dmentioned in discord edgeTx (search for carryTrim))
regards ChrisOhara

3djc, 2023-01-03:
There is something wrong fro sure looking at the code, but I'm not understanding the part where you said it always return 0, it should return un integer (but not write one)

ChrisOhara57, 2023-01-03:
What I meant is:
- the function model.getInput does return an Integer; but the value of which seems to be always "0" no matter what the real value in the input-line is.
- the function model.insertInput does always define an input-line where the value for trim-option is always off, regardless what value you give carryTrim.

If you search in edgeTX for carryTrim - you will find that they discussing the same problem.

I think that the declaration of the function has been "copied" from the mixer definitions.
There is a trim (yes/no) definition there which makes sense to be a boolean.
But in inputs there is a choice of different options so boolean does make no sense.

3djc, 2023-01-03:
> * the function model.getInput does return an Integer; but the value of which seems to be always "0" no matter what the real value in the input-line is.

That's where I'm puzzled, the code should work ok for that (insert definitely is flawed)

ChrisOhara57, 2023-01-03:
Sorry, you are right
return values are:
on = 0
off = 1
Rud = -1
Ele = -2
Thr = -3
Ail = -4

3djc, 2023-01-03:
Ok, the insert is an easy fix then

ChrisOhara57, 2023-01-03:
glad to hear.

(But, please keep in mind, that the description in the reference manual is not correct.
And the name carryTrim does suggest something different (at least in my German-English)
Also negative values are not what I would have expected.)

3djc, 2023-01-04:
Could you link the edge discussion ? I cannot seem to be able to find it

rotorman, 2023-01-04:
Probably following discussion in EdgeTX Discord is meant here: https://discord.com/channels/839849772864503828/842693696629899274/1049103335686406195

3djc, 2023-01-04:
Well coming later to a discords discussion is a nightmare, so lets keep relevant discussion and info here, I'll fix edge too.

From where I stand:
model.getInput works fine. We could debate forever if internal data should have been transformed, but since it has been live like that for a long time, I think changing now would be counter productive and break existing scripts.

model.insertInput needs fixing to allow interger values.

I also agre that carryTrim is a bad name, but for script compatibility we cannot remove it, so what we could do is deprecate it, and also allow a new name in addition to carryTrim

ChrisOhara57, 2023-01-04:
That's ok.
The only important thing is that model.insertInput works.
The rest is trivial.
Thanks

3djc, 2023-01-06:
@ChrisOhara57 , what firmware do you run ? I would like to make a test version for you

ChrisOhara57, 2023-01-06:
I do run 2.3.15  Taranis X9lite S

[image]

3djc, 2023-01-06:
Try with this

[x9lites-fix-carryTrim.zip](https://github.com/opentx/opentx/files/10360235/x9lites-fix-carryTrim.zip)

You can use deprecated carryTrim and get negative values, or new trimSource and get positive ones. Please reports any issue here

ChrisOhara57, 2023-01-06:
**seems to work!**

Here are only a few parts of the code I used

	    lineAmount = model.getInputsCount(inputNumber);
	    if (lineAmount == 1) then
		inputData = model.getInput(inputNumber, 0 );
			if (inputData.source == 78) then           --###### Aileron
			      inputData.source = 203;
			      inputData.carryTrim = -4;
			      model.insertInput(inputNumber, 1, inputData);
			      model.deleteInput(inputNumber, 0);

read an input line,  changes some data and inserted it successfully as a new input line.

seems to be ok !!!!

**Thanks** a lot (very fast response)

3djc, 2023-01-06:
Safe to use,new version is not anytime soon

ChrisOhara57, 2023-01-06:
OK!

But I do have a problem at the moment.
With the test Firmware the script worked for a few test.
And now suddenly it looks like it's the same as before.
I can not explain what I did. It looks like I lost the modification.
I'm sure it's my fault but I cannot find the reason.

ChrisOhara57, 2023-01-06:
update just FYI

my scripts work with trimSource (positiv values).

carryTrim is ignored  (positiv or negativ values, tried both)
but I am sure that did work first two test with different models

but that's ok for me

ChrisOhara57, 2023-01-10:
update just FYI
did some more tests with new firmware

----------
First I used only one object "inputData" to read the table, changed carryTrim and wrote it back as a new inputline.
Most of the time the value for carryTrim is only "0" (instead of -4 as example)
It looks like the behavier depends on the fact what value the original value is.
If it is "0" and I change it to -4 it is not stored.

local inputData = { name, inputName, source, weight, offset, switch, curveType, curveValue, carryTrim, flightModes}
  inputData.carryTrim = -4;
  model.insertInput(inputNumber, 1, inputData);

----------
If I do it this way.
Read the inputData and copy every value into newData (and newData.carryTrim=-4) it does always work.

local inputData = { name, inputName, source, weight, offset, switch, curveType, curveValue, carryTrim, flightModes}
local newData = { name, inputName, source, weight, offset, switch, curveType, curveValue, trimSource, flightModes}
  inputData = model.getInput(inputNumber, 0 );
  newData.name = inputData.name;
     copy all values this way exept carryTrim (or trimSource)
  newData.carryTrim = -4;
  model.insertInput(inputNumber, 1, newData);

----------

It's the same when I use trimSource instead of carryTrim.

3djc, 2023-01-10:
yes, there is an issue with the proposed change when carryTrim and trimSource define different values. Will work on fixing that

3djc, 2023-01-10:
Could you try this ? carryTrim is removed completely, use trimSource and positive values

[x9lites-test-trimSource.zip](https://github.com/opentx/opentx/files/10380870/x9lites-test-trimSource.zip)

ChrisOhara57, 2023-01-10:
> 

I'm not sure whether you understood.
I meant that even when I use carryTrim only for both getinput and insertInput it does not work when the original value of carryTrim is "0" (what means "ON").
I assume that probably LUA "thinks" it's a real boolean value and so it defines inputData.carrytrim as a real Boolean and this may cause a problem when calling insertInput (which expects a number declaration)???

At the moment I do not have a problem.

But anyway I will test the new firmware  and will give a feedback.

3djc, 2023-01-10:
Before the last firmware, the read would get you both carryTrim and trimSource, then when you apply it, both will be submitted, producing unreliable results (last value processed wins)

ChrisOhara57, 2023-01-10:
> Could you try this ? carryTrim is removed completely, use trimSource and positive values
> 
> [x9lites-test-trimSource.zip](https://github.com/opentx/opentx/files/10380870/x9lites-test-trimSource.zip)

Cannot test. After flashing the firmware the radio does not recognize the SDcard anymore.
After flashing back to previous version it works again.

3djc, 2023-01-11:
You probably had an issue with your flash, could you retry ? There is no way sd would be impacted by that change

ChrisOhara57, 2023-01-11:
It does work. Tried three models (two brand new and a copy of an existing one and it worked well.

(PS: it was not the download. Copying firmware to radio was the problem. I needed a new SDcard)

PR #8997 (2023-09-21): Fix carryTrim in LUA
- rename the  carryTrim in ExpoData to trimSource, and fix it's lua casting
- trimSource is using positive values, where carryTrim was using negative ones
- this fixes #8995
closed 2023-09-21 by bsongis