opentx/opentx #9006: 2.2.4 SDCard content

lorenzo61, 2023-01-26:
Hello, 
I would like to report that the link (https://downloads.open-tx.org/2.2/release/sdcard/ ) to download the content of SDCard for the release 2.2.4 (2.2V0021) is unavailable
In the index of /2.2/release/ there is not a directory sdcard like the other releases.

Lorenzo

HobbieJoe, 2023-06-12:
Hi, my X9D+ failed to bind to S8R with ACCST 2.1.0 found in software update 2.3.15.
I cannot open BIN file for 2.2.4 in OpenTX since firmware upgrade. So, cannot work backwards from upgrade. Cannot bind S8R nor X8R, just solid green on attempts to bind. John

JFingerle, 2023-09-06:
Please close this outdated issue. Thank you!

3djc, 2023-09-06:
Those will be catch by the cleaning mechanism

constantdupuis, 2024-05-16:
Hello, I have the same issue, I would like to update my Taranis with OpenTX 2.2.4, but sdcard content link is broken, any other way to get this content ?

Thanks