opentx/opentx #9009: External RF "R9M Lite Pro" not working

vbond-app, 2023-02-23:
**Describe the bug**
I have Jumper T16 Pro with External RF R9M Lite Pro.
Jumper T16 Pro has been hardware modified for support R9M ACCESS.
But i cannot register external receivers.
Reciever FrSky R9 SX.

**To Reproduce**
Steps to reproduce the behavior:
1. go to radio settings -> external module -> select R9M  ACCESS -> press REGISTER;
2. Nothing happens during the register process; (Waiting ...)

Expected behavior
The expected behavior is a binding between receiver and radio module

**General information**
 - OpenTX version 2.1.15 Jumper T6 PRO  with Externalaccessmod Option
 - Hardware  R9M Lite Pro + FrSky R9 SX (latest firmware)

3djc, 2023-02-23:
R9M access is for the access version of 2019 full size. Not sure the lite pro choice is available for your radio

vbond-app, 2023-02-23:
Do the R9M Lite pro suuport?
In the menu section "Modules/RX Versions"  field External information  : "---------------"

vbond-app, 2023-02-23:
I did everything possible
And checked all nearest smd components

3djc, 2023-02-23:
If your radio doesn't offer r9m lite pro choice, then it won't work

vbond-app, 2023-02-23:
Could u say exactly - do opentx 2.3.15 support R9M lite pro?
The latest version had this comment about the changes: "   RADIOMASTER ZORRO
Support for ELRS and multi version (ELRS require choosing internalELRS in companion options)
Support for small size Access modules (R9M pro,…) "

3djc, 2023-02-23:
But you don't have a sorry there, or do I miss something?

vbond-app, 2023-02-23:
I have Jumper t16 , I think that u soft should support rf module lite too ?
about protocol , look , the same : https://www.frsky-rc.com/product/r9m-lite-pro/

3djc, 2023-02-23:
The dialog between the radio and the module is different between almost all r9 modules. From memory, we only support small modules on radio equipped with a small module

vbond-app, 2023-02-23:
I have r9m lite pro
I think , that u only support large modules 
Could show me code/file o with process of detect radio ?
i want change this

closed 2023-02-23 by 3djc

vbond-app, 2023-02-23:
It is not solved yet

3djc, 2023-02-23:
We don't support the radio/module configuration, so it is not a bug